Xinyuan-LilyGO/T-Display-S3 #2: TFT example build fails with PlatformIO

hpsaturn, 2022-08-01:
## Overview

Into the current tft directory sample, pio fails with this output:

```cpp
(main) avp:tft$ pio run
Warning! Ignore unknown configuration option `board.psram_type` in section [env:ESP32-S3-DevKitC-1]
Warning! Ignore unknown configuration option `board.code_debug` in section [env:ESP32-S3-DevKitC-1]
Warning! Ignore unknown configuration option `board.memory_type` in section [env:ESP32-S3-DevKitC-1]
Processing ESP32-S3-DevKitC-1 (platform: espressif32; board: ESP32-S3-DevKitC-1; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Unknown board ID 'ESP32-S3-DevKitC-1'
```

LilyGO, 2022-08-02:
I don't get this error. Warning should not affect the compilation of the program

hpsaturn, 2022-08-02:
In my Debian Linux, PlatformIO doesn't compile it with this case sensitive issue

closed 2022-08-03 by LilyGO