Xinyuan-LilyGO/T-Display-S3 #4: Factory sample fails on PlatformIO build

hpsaturn, 2022-08-01:
## Overview

Adding the fix from the PR #3 in this sample, for fix similar issue than #2, I have the next output:

```cpp
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.2% (used 39972 bytes from 327680 bytes)
Flash: [======Error: The program size (1323221 bytes) is greater than maximum allowed (1310720 bytes)
*** [checkprogsize] Explicit exit, status 1
====]  101.0% (used 1323221 bytes from 1310720 bytes)
```

LilyGO, 2022-08-02:
It should be that the animation or font takes up a lot of space, resulting in insufficient flash space. In this case, the partition table can be modified to expand the space for storing programs.

hpsaturn, 2022-08-02:
I tested it with two partitions schemas but I had the same error.

LilyGO, 2022-08-03:
Maybe there is still a problem with the configuration of platfromIO.

hpsaturn, 2022-08-03:
I will submit a Github action for try to probe this hypothesis

LilyGO, 2022-08-03:
I am reconfiguring platfromIO. ini

LilyGO, 2022-08-03:
I have reconfigured the platfromIO.ini file in the factory

hpsaturn, 2022-08-03:
From the continuous integration workflow (with my last PR #6 ) sound like that the compiling issue was resolved. I'm out of my house, I don't have here the board, but sounds like that will works.

closed 2022-08-03 by hpsaturn

Jason2866, 2022-10-06:
Why are you using still the outdated Arduino Core 2.0.3? Many bug fixes have been done in actual Core 2.0.5.