Xinyuan-LilyGO/T-Display-S3 #7: Can't connect to ttyACM0

wdevore, 2022-08-12:
Hi, I plugged in board and it starts with the LilyGO example. I tried to upload the GetChipID but it won't connect and times out. I added a udev rules for ttyACM0:
```KERNEL=="ttyACM*", ATTRS{idVendor}=="303a", ATTRS{idProduct}=="1001", ENV{ID_MM_DEVICE_IGNORE}="1"```
but it still won't connect in Arduino. I have /dev/ttyACM0 selected for the port.

This is running the get_idf
```
Setting IDF_PATH to '/home/iposthuman/esp/esp-idf'
Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool xtensa-esp32-elf found in PATH: esp-2020r2-8.2.0.
Not using an unsupported version of tool xtensa-esp32s2-elf found in PATH: esp-2020r2-8.2.0.
Not using an unsupported version of tool openocd-esp32 found in PATH: v0.10.0-esp32-20191114.
Using Python interpreter in /home/iposthuman/.espressif/python_env/idf4.3_py3.8_env/bin/python
Checking if Python packages are up to date...
Python requirements from /home/iposthuman/esp/esp-idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /home/iposthuman/esp/esp-idf/components/esptool_py/esptool
  /home/iposthuman/esp/esp-idf/components/espcoredump
  /home/iposthuman/esp/esp-idf/components/partition_table
  /home/iposthuman/esp/esp-idf/components/app_update
  /home/iposthuman/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin
  /home/iposthuman/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin
  /home/iposthuman/.espressif/tools/xtensa-esp32s3-elf/esp-2020r3-8.4.0/xtensa-esp32s3-elf/bin
  /home/iposthuman/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin
  /home/iposthuman/.espressif/python_env/idf4.3_py3.8_env/bin
  /home/iposthuman/esp/esp-idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
```
The Arduino terminal prints the Wifi connection info but I can't program it with the GetChipID example:
```
scan done
4 networks found
1: xxxxx (-64)*
2: yyyyy (-89)*
3: zzzzzzz (-91)*
4: bbbbb (-93)*

Connecting to xinyuandianzi.......................
```
I have a TTGO T-Display that is working fine.

What do i do? Thanks.

LilyGO, 2022-08-13:
Please let me know if your configuration is the same as the image below.

[image]

wdevore, 2022-08-13:
Yes. That is the configuration I chose.

LilyGO, 2022-08-15:
Please use arduino's ESP32, version 2.0.3 to test this code. Or execute the simplest blink program.

[](https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/example/arduino/factory)

LilyGO, 2022-08-15:
https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/example/arduino/factory

Tissy, 2022-08-22:
I had to unplug the device from the USB connector, hold down the RST button on the S3, whilst keeping button help down, reconnect USB and upload code.  That resolved a connection issue for me.

wdevore, 2022-08-22:
Cool. I'll give that a try :+1:

wdevore, 2022-08-25:
Holding the RST pin while connecting it didn't work. It won't stop crashing. I guess it is bricked. I don't windoz so I can't run a firmware update in the hopes that would un-brick it.

LilyGO, 2022-08-25:
You can disconnect the USB connection first, then press and hold the boot button to connect the USB.

wdevore, 2022-08-25:
Holding "Boot" button down does seem to stop it from crashing. But when I try to upload to dev/ttyACM0 it fails with

in my sdkconfig file i'm using:
```
CONFIG_IDF_TARGET="esp32s3"
CONFIG_IDF_TARGET_ESP32S3=y
```
I then try an flash it:
```
$ idf.py -p /dev/ttyACM0 flash

Executing action: flash
Running ninja in directory /media/iposthuman/Nihongo/Hardware/Esp32/TTGO-Espressif-Projects/blink/build
Executing "ninja flash"...
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cd /home/iposthuman/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PAT...cts/blink/build" -P /home/iposthuman/esp/esp-idf/components/esptool_py/run_esptool.cmake
esptool.py --chip esp32s3beta2 -p /dev/ttyACM0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x0 bootloader/bootloader.bin 0x10000 blink.bin
esptool.py v3.0
Serial port /dev/ttyACM0
Connecting...
WARNING: This chip doesn't appear to be a ESP32-S3(beta2) (chip magic value 0x00000009). Probably it is unsupported by this version of esptool.
Chip is ESP32-S3(beta2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 00:00:00:00:00:00
Uploading stub...
Running stub...

A fatal error occurred: Invalid head of packet (0x47)
CMake Error at run_cmd.cmake:14 (message):
  esptool.py failed
Call Stack (most recent call first):
  run_esptool.cmake:21 (include)

FAILED: CMakeFiles/flash 
cd /home/iposthuman/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH="/home/iposthuman/esp/esp-idf" -D ESPTOOLPY="/home/iposthuman/.espressif/python_env/idf4.3_py3.8_env/bin/python /home/iposthuman/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32s3beta2" -D ESPTOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="/media/iposthuman/Nihongo/Hardware/Esp32/TTGO-Espressif-Projects/blink/build" -P /home/iposthuman/esp/esp-idf/components/esptool_py/run_esptool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

Maybe my configuration is wrong?

LilyGO, 2022-08-26:
Try to erase using the command if it was successful.` idf.py -p /dev/ttyACM0 erase-flash`

wdevore, 2022-08-27:
I re-installed espressif SDK and made sure I installed esp32,esp32s2,esp32s3
It doesn't crash anymore, but doesn't seem to work either. The monitor show "waiting for download" --forever.
I tried the hello_world too, but I get the same monitor message.
If I click the Reset button it goes back to crashing. This thing is BRICKED!

I did the following:
- cd ~esp/esp-idf/examples/get-started/blink
- idf.py set-target esp32s3
- idf.py menuconfig
  - Set LED type to GPIO
  - Set LED pin to gpio 21     <-- connected Oscilloscope to pin
  - Set blink period to 100ms
- idf.py build
- idf.py -p /dev/ttyACM0 flash

it finally flashed--without crashing (see further below), however, it doesn't toggle gpio 21.
The Green LED is solid, the Red flickers constantly.

When I connect using the monitor: ```idf.py -p /dev/ttyACM0 monitor``` I get the following output:
```
--- idf_monitor on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x0 (DOWNLOAD(USB/UART0))
Saved PC:0x40041a7c
waiting for download
```

Flash output:
```
...
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 68:b6:b3:21:7c:bc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00004fff...
Flash will be erased from 0x00010000 to 0x00042fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 20288 bytes to 12837...
Writing at 0x00000000... (100 %)
Wrote 20288 bytes (12837 compressed) at 0x00000000 in 0.4 seconds (effective 418.2 kbit/s)...
Hash of data verified.
Compressed 207456 bytes to 107625...
Writing at 0x00010000... (14 %)
Writing at 0x0001c6c8... (28 %)
Writing at 0x0002213c... (42 %)
Writing at 0x000284ff... (57 %)
Writing at 0x0002ed20... (71 %)
Writing at 0x00039244... (85 %)
Writing at 0x0003f1d2... (100 %)
Wrote 207456 bytes (107625 compressed) at 0x00010000 in 2.1 seconds (effective 772.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 505.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
```

LilyGO, 2022-08-27:
Because using internal USB not UART. If the program is uploaded successfully for the first time, it needs to be powered on again. If the monitor still shows waiting for download, there may be a problem with the BOOT button or the line.

wdevore, 2022-08-27:
If I power it back on --without holding the Boot button-- it returns to constantly crashing. The monitor just shows it repeatedly crashing.

bigsun111, 2022-09-08:
I experienced the same problem using arduino's ESP32, 2.0.4. Solved it by switching back to 2.0.3. Not sure what triggers it.

rimim, 2022-09-13:
The issue is in the 2.0.4 board package. To use 2.0.4 replace ".arduino15/packages/esp32/tools/esptool_py/3.3.0/esptool.py" with ".arduino15/packages/esp32/hardware/esp32/2.0.4/tools/esptool.py". Not sure why they have two copies of esptool, but the one in the 2.0.4 directory is the correct version the other one is older.

Jason2866, 2022-10-06:
Tasmota does now support this board too. Nice LVGL performance.

mmMicky, 2022-12-02:
Problem seems to be resolved. I will close this issue

closed 2022-12-02 by mmMicky