Xinyuan-LilyGO/T-Display-S3 #8: 16MB flash setting not working

erongd, 2022-08-16:
The board clearly has 16MiB flash, however the given  arduino examples only work with 8MiB. When I manually change to 16MiB in platformio.ini the board report flash error on boot.

LilyGO, 2022-08-17:
I just used the Arduino to choose a 16MB flash size with a 16MB partition. Everything works fine.

erongd, 2022-08-17:
I use platformio with vscode. Here is my setting in platformio.ini for 16MB flash.

board_build.partitions = default_16MB.csv
board_build.arduino.memory_type = qspi_opi
board_build.flash_size = 16MB
board_build.psram_type = opi

The board went into bootloop with message as follows.

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x43c
load:0x403b6000,len:0xbd0
load:0x403ba000,len:0x29c8
SHA-256 comparison failed:
Calculated: 0bb45b01207a4a9e9fca8880ddfc0f79112bfa4d004bd1e6b2eaa75c5323b19d
Expected: 6eb14ec480aff4f05e39378c41406102ac163573741e63b6cfd73509ef44e3bd
Attempting to boot anyway...
entry 0x403b61d8
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x403bb22e

erongd, 2022-08-17:
It seems the factory example works in ArduinoIDE even with arduino_esp32 2.0.4. But not with platformio for some reason.

erongd, 2022-08-18:
I got it working with platformio. I switched the board to "esp32s3box" which has 16MB built into it. I also changed memory_type to "qio_opi" which is what arduino_esp32 2.0.4 has.

The factory example works with platformio on vscode with the latest 2.0.4 arduino_esp32.

Here is my configuration for platformio.ini.
[env:esp32s3box]
platform = espressif32
board = esp32s3box
framework = arduino

upload_speed = 921600
monitor_speed = 115200

board_build.arduino.memory_type = qio_opi
board_build.psram_type = opi

board_build.flash_mode = qio
board_build.flash_size = 16MB
board_build.partitions = ffat.csv

closed 2022-08-18 by erongd

LilyGO, 2022-08-22:
I guess esp32s3box and esp32s3dev should need to be modified and added in some parameters. But at the moment I haven't looked into the parameters it needs.
thanks for sharing.

handledexception, 2022-09-25:
I've submitted a PR to the PlatformIO project to add a board config .json file for the ESP32-S3-DevKitC-1-N16R8V board that should fix this problem.
https://github.com/platformio/platform-espressif32/pull/921

psm12, 2022-12-21:
I am having issues just getting the serial monitor working on this board with platformIO. It seems inconsistent and although I got the demo working (teastain)I have not got a simple 'hello world' demo to work or the t-display-s3.   I'm not sure if it is the lilygo dev board, the espressif s3 chip, platformio or my inexperience or inept inability that is the problem. Does anyone have a definitive Platformio.ini that works with this dev board?
(With apologies for high jacking the thread)
Paul

teastainGit, 2022-12-22:
Hi Paul, 
You could try out my “admittedly Arduino” setup repo, here:

https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples￼
teastainGit/LilyGO-T-display-S3-setup-and-examples: LilyGO T-Display S3 setup and examples for Arduino IDE
github.com
 
Try this Arduino Board selection option:

“ESP32S3 Dev Module"
Upload Speed: "921600"
USB Mode: "Hardware CDC and TAG"
USB CDC On Boot: "Enabled"
USB Firmware MSC On Boot: "Disabled"
USB DFU On Boot: "Disabled"
Upload Mode: "UARTO / Hardware CDC'
CPU Frequency: "240MHz (WiFi)"
Flash Mode: "QIO 80MHz"
Flash Size: "16MB (128Mb)"
Partition Scheme: "Huge APP (3MB No OTA/1MB SPIFFS)"
Core Debug Level: "="
PSRAM: "OPI PSRAM"
Arduino Runs On: "Core 1"
Events Run On: "Core 1"

Cheers !
Terry Stainton

***@***.***

psm12, 2022-12-26:
Thanks Terry, 
That was very helpful. 
Paul