Xinyuan-LilyGO/T-Display-S3 #9: Cannot get display to work

Tissy, 2022-08-22:
Hi,

I am trying to run - https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/example/arduino/tft

I have updated the TFT_eSPI library and using Arduino IDE, however not getting anything on the display at all.

What line should be uncommented in User_Setup_Select.h please for this display to work.

Many thanks,

lyusupov, 2022-08-25:
> What line should be uncommented in User_Setup_Select.h please for this display to work.

https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/lib/TFT_eSPI/User_Setup_Select.h#L52

Blastaar-BR, 2022-09-28:
I am having same issue. Used the lib from the manufacturer github ( https://github.com/Xinyuan-LilyGO/T-Display-S3?spm=a2g0o.detail.1000023.1.48c84641mSoXSV ), the line #52 is already uncommented, using  Arduino IDE 2.0, compiler works fine, code is uploaded to the board but it remains in black screen. Any other hint to fix it?

Blastaar-BR, 2022-09-29:
Hi, now I got it working, but only with the version 1.8.12 of Arduino IDE. With version 2.00 I was not able to make it working, don´t know why. I followed the following great video from Volos Project. He also uses V. 1.8.12:  https://www.youtube.com/watch?v=gpyeMjM9cOU&t=338s
@LilyGO please investigate what is wrong with V2.00 of Arduino IDE! Tks

LinuxSBC, 2022-10-02:
Arduino IDE V2 works perfectly for me. My guess is that you needed to use the TFT_eSPI library provided by LilyGO.

jessepdx, 2022-10-03:
i moved back to 1.8.19 and it works

2.0 seems to overwrite the TFT_eSPI library

LilyGO, 2022-10-06:
I haven't tested with arduino 2.0.0. But my guess is that TFT_eSPI was updated or overwritten. Please use the TFT_eSPI library provided in the repository to compile.

robertlugg, 2022-10-07:
It works for me.  Remember after you upload and get a blank screen to press the RST button on the board and see.

Blastaar-BR, 2022-10-07:
Now I got the V2.00 working. The problem is in the TFT_eSPI library different than V2.4.71 (supplied by the manufacturer). Please refer to this link here:  

[https://forum.arduino.cc/t/arduino-ide-2-00-and-lilygo-t-display-esp-s3-black-screen/1036875/6](url)

LilyGO, 2022-10-13:
The problem has been pointed out. Close the question.

closed 2022-10-13 by LilyGO