Xinyuan-LilyGO/T-Display-S3 #12: Where is the schematic for this board

DrJaymz, 2022-09-19:
I haven't found the schematic anywhere.

HrOlsen, 2022-09-27:
it's highly appreciated if you add the schematic

DrJaymz, 2022-09-27:
> it's highly appreciated if you add the schematic

I agree, because when we're trying to get extra low power standby current its important to see the schematic.

HrOlsen, 2022-10-11:
thanks for adding the schematic to the project

DrJaymz, 2022-10-11:
ok, I'm looking at the schematic, I'll need a moment to get my head around what that is showing.  In particular, I'm looking to see what parasitic loads there is on the battery when we're in deep sleep.

So there's 200k load on BAT bus to ground thats 20uA to start with with 4V battery.  
Then I can't quite work out the connection between VBAT and BAT buses.  But looks like VBAT is loaded with another 100K R12 which is another 40uA.  So thats 60uA.
Then there's RED LED8 dangling off the 3V3 bus.  My actual board has a bright green LED next to the USB C. is that this one?  I thought it may not be active but I just tried with a battery and yes its on, burning away a whopping 5mA all the time.

So, you can make some nice things with a standby current of <100uA, single button power etc.  But not with that LED knocking about.

LilyGO, 2022-10-13:
The T-Display-S3 without touch when entering deep sleep is about 200uA.

rimim, 2022-10-23:
Drive GPIO15 (PWR_EN) low and it will shut off the green LED. 

`
#define PIN_POWER_ON           15
#define PIN_LCD_BL             38
#define PIN_WAKEUP_FROM_SLEEP  10

void deepSleep()
{
    digitalWrite(PIN_POWER_ON, LOW);
    digitalWrite(PIN_LCD_BL, LOW);

    gpio_num_t wakeUp = gpio_num_t(PIN_WAKEUP_FROM_SLEEP);
    esp_sleep_enable_ext0_wakeup(wakeUp, 0); // 1 = High, 0 = Low
    rtc_gpio_pullup_en(wakeUp);
    rtc_gpio_pulldown_dis(wakeUp);
    esp_deep_sleep_start();
}
`

LilyGO, 2022-10-26:
Problem has been resolved.

closed 2022-10-26 by LilyGO