Xinyuan-LilyGO/T-Display-S3 #13: Factory example fails to compile Espressif releases > 2.0.3

noisysoil, 2022-09-27:
**Issue**
Factory example fails to compile with Espressif Arduino releases > 2.0.3.

**Description**
Your `factory` example as of commit `ebec56d02c5e87aa1f4cf522d27f678849e3dafe` uses the Espressif Arduiono library version 2.0.3, as defined in `platformio.ini`:

`platform_packages =
    framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.3`

You state that it should work above these versions, but the `factory` example project fails to compile with #2.0.4, #2.0.5 or #master, and fails with the following error:

`lv_conf_kconfig.h:15:14: fatal error: sdkconfig.h: No such file or directory`

The LvlGL version tested in `platformio.ini` is:

`lib_deps = 
	lvgl/lvgl@^8.3.2`

**Mitigation**
Use version 2.0.3 but it's not ideal.

LilyGO, 2022-10-13:
It can compile 2.0.4 and 2.0.5.
The reason for the error is because here is modified in the above two versions. https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/c4e22fca21e07d27a1d4df8f262494901ea4d6d2/example/factory/platformio.ini#L34

closed 2022-10-13 by LilyGO

noisysoil, 2022-10-13:
> It can compile 2.0.4 and 2.0.5. The reason for the error is because here is modified in the above two versions.
> 
> https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/c4e22fca21e07d27a1d4df8f262494901ea4d6d2/example/factory/platformio.ini#L34

+1 for the fix, it works - thanks!