Xinyuan-LilyGO/T-Display-S3 #14: Library path Arduino 2.0

Bogdan50, 2022-09-27:
I am having difficulties adding the libraries to the path.
Could you please show how to add those for the new Arduino IDE 2.0?

robertlugg, 2022-09-29:
@Bogdan50 , The `lib` files should be moved to your _sketchbook_ and NOT where the system ones are.  So, on Windows, mine is in .../Documents/Arduino/libraries/
[image]

I still can't upload to the S3 though.  What is the name of the board you are using in the Arduino IDE?

LilyGO, 2022-10-13:
The sample program needs to rely on the files in lib to compile. The way to use it is to copy the files in the lib to the user's Arduino/libraries folder. In Arduino 2.0.0 it is possible to automatically upgrade the version of the library. I have made minor modifications to this TFT library. So there will be a problem that it cannot be used after being upgraded. It is recommended not to upgrade the version of the library or patch the latest board library.

closed 2022-10-13 by LilyGO