Xinyuan-LilyGO/T-Display-S3 #15: Add LED function to board (Multiple options)

robertlugg, 2022-09-28:
There doesn't seem to be an LED so that I might run the classic "blinking LED" demo.  It seems almost every introduction video of many boards use that as a least-common-denominator example.  Maybe a colorful LED that lights up the LILYGO logo under the acrylic would be useful, attractive and also highlight your brand.

Are any of the lights I see programmable (green power, red solid, LCD, LCD backlight)?  If so, can you provide the GPIO Port or perhaps a small code example to turn it on and off.  A blinky example that flashed the entire LCD between black and white might also be a reasonable substitute.

LinuxSBC, 2022-10-02:
I haven't yet gotten around to figuring out the other LEDs, but the backlight is on pin 38. Sub in 38 for LED_BUILTIN in the Arduino IDE Blink example, and it should work. However, it sounds like this is one of your first ever microcontrollers. If so, I would recommend getting something else, like a cheap Arduino Nano or Uno clone, as this board is still quite new and unknown, as well as having many advanced features, so you need to have preexisting knowledge to use it for much of anything.

robertlugg, 2022-10-07:
@LinuxSBC thanks.

closed 2022-10-07 by robertlugg