Xinyuan-LilyGO/T-Display-S3 #16: Device is bricked

Arielhh, 2022-10-05:
The device is not visible in device manager with a com port.
Windows see it as a drive which has inside one file called Firmware.bin

LilyGO, 2022-10-06:
In the case of power failure, press and hold the BOT button and then connect to the computer, the port will reappear and can be used to burn the program.

robertlugg, 2022-10-07:
Keep "playing" with it.  Do you know where the three buttons are on the device.  Two are push buttons between the USB plug and the display.  The third is harder to find.  Its on the side.  Press combinations of those.  I believe the BOOT + RST is what is needed.  Refer to the diagram in the readme for pictures to locate the buttons.

ArielIntel, 2022-10-08:
Thanks it's alive again but when loading the example programs nothing is working.

ArielIntel, 2022-10-08:
> In the case of power failure, press and hold the BOT button and then connect to the computer, the port will reappear and can be used to burn the program.

Now that it's working again the example codes are not working   
I am using the latest arduino IDE.

LilyGO, 2022-10-08:
There will be no response after uploading the program in download mode. User code execution begins only after a reset.

Arielhh, 2022-10-08:
After reset the port disappears and reappears continuously  Accompanied by
Windows dinging every time it discovers the usb port after it disappeared.

LilyGO, 2022-10-08:
That's because the program keeps resetting and constantly enumerating the device. Press and hold the BOT and then click the reset button to manually enter the download mode.

Arielhh, 2022-10-10:
Thanks, your suggestion solved the problem but still the example programs are not working. Do you have a much simple example program that will work?

robertlugg, 2022-10-10:
HI @Arielhh, if you aren't using Arduino IDE, I suggest you switch to it.  The main advantages are that its pretty good and very popular.  That allows more people to help.  I also wanted to mention what someone told me.  And that is this board is new and it takes a while for the community to adopt it.  That's why, for instance, you will see a custom TFT_eSPI library in this github.  That's all to say that if you are getting started and about to give up, go ahead and switch to some more basic ESP32 board.  Select one based on the ones that are used in the tutorials you find.  This t-display S3 is beautiful, and you will figure it out...it just will take more time and frustration, but I am learning a lot by doing something over my head.

In terms of program, the demos in the Arduino work, but require that you change things (at least I needed to).  Get the blink code:
[image]
Notice in the code the symbol LED_BUILTIN ?  There are three locations.  Replace each one with 38:
[image]

Upload it until you see the text "Hard resetting via RTS pin..."  At that point, press the reset button (only) on your t-display S3 and you should see the backlight flash on and off.

Arielhh, 2022-10-10:
Thanks. I am experienced with ESP32 but for some reason have a lot of
problems with this new board. I guess you are right, it takes time to get
it working properly. Thanks again for your help

Arielhh, 2022-10-11:
Done that but after uploading the sample program nothing happens. When I
press the reset button the below cycle starts again.

Arielhh, 2022-10-11:
I managed to upload the sample program. Everything is working fine but the
device is rebooting every few seconds even when i connect it to external
power source  (not via the computer). How can i prevent this
constant reboots?

tyeth, 2022-10-11:
That's usually a crash. Try another example:
https://github.com/teastainGit/LillyGO-T-display-S3-setup-and-examples
I was able to use that and also get my own animated gifs (via movie from powerpoint then ezgif.com) working via 
https://github.com/tyeth/bitbank2-AnimatedGIF/tree/master/examples

Arielhh, 2022-10-12:
> That's usually a crash. Try another example: https://github.com/teastainGit/LillyGO-T-display-S3-setup-and-examples I was able to use that (plus custom background and VEML7700 sensor) and also, separately get my own animated gifs (via movie from powerpoint then ezgif.com) working via https://github.com/bitbank2/AnimatedGIF/tree/master/examples/TFT_eSPI_memory

Thanks :)

makerming1, 2022-11-30:
> 

Thanks a lot!!!! Your documentation save my life!

makerming1, 2022-11-30:
> That's usually a crash. Try another example: https://github.com/teastainGit/LillyGO-T-display-S3-setup-and-examples I was able to use that (plus custom background and VEML7700 sensor) and also, [...]

Thanks a lot!!!! Your documentation save my life!

makerming1, 2022-11-30:
> HI @Arielhh, if you aren't using Arduino IDE, I suggest you switch to it. The main advantages are that its pretty good and very popular. That allows more people to help. I also wanted to mention [...]
Thanks for providing this simple demo! I was able to upload the code and see how it works the first time!

closed 2023-01-06 by mmMicky