Xinyuan-LilyGO/T-Display-S3 #18: Tasmota support

Jason2866, 2022-10-06:
We added today support for this device. Nice LVGL performance.

Greetings from Tasmota crew.

LilyGO, 2022-10-13:
thank you for your feedback.

closed 2022-10-13 by LilyGO

Scanboostar, 2023-01-03:
Is there documentation how to setup the [T-Display-S3](https://github.com/Xinyuan-LilyGO/T-Display-S3) with Tasmota. I can't figure out how to configure the display. Tried  tasmota32s3-display.factory.bin with Lilygo-S3-Display.autoconf.

ianmxyz, 2023-01-06:
I'm unable to get this to flash Tasmota, tried various ways and eventually using the updated Esp-flasher seemed to work, except the device is stuck in a bootloop. 

> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
> Saved PC:0x403bb0ce
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fcd0108,len:0x39c
> load:0x403b6000,len:0x9a4
> load:0x403ba000,len:0x2868
> SHA-256 comparison failed:
> Calculated: 96edcbf24307eec974a15a9c36833138c4d40b02a8c0319c7b22c9c6b2fd11eb
> Expected: 867c0d6cb8d08f5a22bebcb1e4a5d457c2700a1c6b76abeb4d778eadcc3446e2
> Attempting to boot anyway...

Scanboostar, 2023-01-06:
> 
I use esptool to flash the [T-Display-S3](https://github.com/Xinyuan-LilyGO/T-Display-S3).
Command: 
`esptool --chip esp32-s3 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode keep --flash_size keep 0x0 tasmota32s3-lvgl.factory.bin`

Jason2866, 2023-01-06:
Or easy way use Tasmota Web Installer https://tasmota.github.io/install/

ianmxyz, 2023-01-06:
> Or easy way use Tasmota Web Installer https://tasmota.github.io/install/

That's what I had tried first . . .

ianmxyz, 2023-01-06:
> > 
> 
> I use esptool to flash the [T-Display-S3](https://github.com/Xinyuan-LilyGO/T-Display-S3). Command: `esptool --chip esp32-s3 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode keep --flash_size keep 0x0 tasmota32s3-lvgl.factory.bin`

Thanks, I'll try with those settings.

ianmxyz, 2023-01-07:
> > 
> 
> I use esptool to flash the [T-Display-S3](https://github.com/Xinyuan-LilyGO/T-Display-S3). Command: `esptool --chip esp32-s3 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode keep --flash_size keep 0x0 tasmota32s3-lvgl.factory.bin`

Sadly this didn't work. I could use it to flash an esphome binary though, and with the correct libraries the Arduino IDE works perfectly too. I was unable to find the specific binary you mention, my presumption was that tasmota32-lvgl.factory.bin is now compatible with the S3 now. I had also tried the plain S3 binary which also doesn't work however.

ianmxyz, 2023-01-07:
If anyone else is as stupid as me, the correct firmware is here. In the end I used the esphome web installer.

[https://github.com/Jason2866/Tasmota-specials/blob/firmware/firmware/tasmota32/other/tasmota32s3-lvgl.factory.bin](url)

Jason2866, 2023-01-07:
@ianmxyz Do **NOT** use any EspHome Installer for flashing Tasmota. You will run in issues.
Tasmota has its own Web Installer. You linked to my special repo and there is the web [installer](https://jason2866.github.io/Tasmota-specials/) too which has to be used.
But every version is [here](https://tasmota.github.io/install/) and the web [installer](https://tasmota.github.io/install/) too

jkrijger67, 2023-01-08:
Hello, 
I used the Installer you advised but in any way I never get a Box to configure WLAN. After Install the device it is never seen as AP. It doesn't matter wich version I use. I tried 2 lilygo s3.

Scanboostar, 2023-01-08:
> Hello, I used the Installer you advised but in any way I never get a Box to configure WLAN. After Install the device it is never seen as AP. It doesn't matter [which] version I use. I tried 2 lilygo s3.

Please try a power supply that can deliver more current.
I just tried with the _development_ Tasmota32 LVGL (English) on ESP32-S3

On the USB port of my PC **no** Ap, using a Raspberry PowerSupply is presented de AP.

jkrijger67, 2023-01-08:
Hello, 
I tried powersupply and both lilygo are presenting AP immidiatly. 
Do you know about a release I can upload to use the display of the lilygo s3?
I am reading about ini-files and config-files but I don't know how to use them all together.

Thanks Johannes.

Scanboostar, 2023-01-08:
If you did a autoconfig of "Lilygo-S3-Display" you can upload the autoexec.be from https://tasmota.github.io/docs/LVGL_in_10_minutes/ . I did not progress beyond that point yet.......

Jason2866, 2023-01-08:
@jkrijger67 goto https://tasmota.github.io/install/ and select from unofficial variants 
[image]
After it has installed goto Tasmota Autoconfig and select  T-Display-S3

Scanboostar, 2023-01-08:
@Jason2866, thanks! I now can run the https://tasmota.github.io/docs/HASPmota/#quick-tour (because there is PSRAM support in this firmware)

tyeth, 2023-01-08:
@Jason2866 Thanks for the mention. Is the Touch variant supported too?

jkrijger67, 2023-01-08:
thank you all for support.
I installed adviced release and it starts up. Wlan config done as well. I am using "Lilygo-S3-Display" for auto-config. "T-Display-S3" was not shown. But Display is not showing anything. I tried dipslaytext but nothing is shown. 
Johannes.

Jason2866, 2023-01-08:
Yes, touch is supported. 
Display shows after powering on for a short time the Tasmota logo.
If you see nothing please check if in Web Frontend the Button shows "on".
If you still see nothing go in Tasmota Web console and type `displaydimmer 100` and hit return. This increases the Back light to full brightness.

jkrijger67, 2023-01-08:
Hello Jason2866, 
thank you for your patience. 
Tasmota logo is shown at startup.
The display command shows: 
{"Display {"Model":17,"Type":0,"Width":320,"Height":170,"Mode":0,"Dimmer":100,"Size":1,"Font":2,"Rotate":0,"Invert":0,"Refresh":2,"Cols":[16,8],"Rows":2}}
For me it all looks fine. But "DisplayText Hallo" doesn't show anything. :-(

jkrijger67, 2023-01-08:
Okay, I lied to you. There is something on display but it is very dark. I attached a picture.
[image]

Jason2866, 2023-01-08:
So do `displaydimmer 100` in the Tasmota console. Time to read Tasmota docu :-)
It is working.

richardsg307, 2023-12-13:
I got my Lilygo T-Disdplay-S3 going using the Web Tasmota flasher with firmware Tasmota32-S3 (QIO Flash/OPI PSRAM). Then I auto-configured with Lilygo S3 Display.
I've been messing trying to get it to display some sensor values, learning as I go but currently stuck not knowing how to clear the old text before writing updated values every 5 seconds. Help!
Here's my Berry code so far...
```
#- start LVGL and init environment -#
lv.start()
import string
import json
tasmota.cmd("DisplayDimmer 50")

hres = lv.get_hor_res()       # should be 320
vres = lv.get_ver_res()       # should be 240 (mine is 170)
#print(hres, vres)
scr = lv.scr_act()            # default screean object
f20 = lv.montserrat_font(20)  # load embedded Montserrat 20

#- Background with a gradient from black #000000 (bottom) to dark blue #0000A0 (top) -#
scr.set_style_bg_color(lv.color(0x0000A0), lv.PART_MAIN | lv.STATE_DEFAULT)
scr.set_style_bg_grad_color(lv.color(0x000000), lv.PART_MAIN | lv.STATE_DEFAULT)
scr.set_style_bg_grad_dir(lv.GRAD_DIR_VER, lv.PART_MAIN | lv.STATE_DEFAULT)

#- Upper state line -#
stat_line = lv.label(scr)
#def t() var data = json.load(tasmota.read_sensors()) var t = data['DS18B20-1']['Temperature']
#print("Booh!")
#stat_line.set_text(string.format("Temp = %.2f", t)) end
if f20 != nil stat_line.set_style_text_font(f20, lv.PART_MAIN | lv.STATE_DEFAULT) end
stat_line.set_long_mode(lv.LABEL_LONG_SCROLL)                                        # auto scrolling if text does not fit
stat_line.set_width(hres)
stat_line.set_align(lv.TEXT_ALIGN_LEFT)                                              # align text left
stat_line.set_style_bg_color(lv.color(0xD00000), lv.PART_MAIN | lv.STATE_DEFAULT)    # background #000088
stat_line.set_style_bg_opa(lv.OPA_COVER, lv.PART_MAIN | lv.STATE_DEFAULT)            # 100% background opacity
stat_line.set_style_text_color(lv.color(0xFFFFFF), lv.PART_MAIN | lv.STATE_DEFAULT)  # text color #FFFFFF
stat_line.set_text(string.format("HRES=%3i, VRES=%3i", hres, vres))
#tasmota.set_timer(5000, t)
#stat_line.set_text("Tasmota")
#stat_line.set_text("hres")
stat_line.refr_size()                                                                # new in LVGL8
stat_line.refr_pos()                                                                 # new in LVGL8

#- Sensor reports -#

def f()
var data = json.load(tasmota.read_sensors())
var oflabel = lv.label(scr)
if f20 != nil oflabel.set_style_text_font(f20, lv.PART_MAIN | lv.STATE_DEFAULT) end
oflabel.set_text("")
oflabel.set_text(string.format("Outflow = %.2f°C", data['DS18B20-1']['Temperature']))
oflabel.set_style_text_color(lv.color_hex(0xffffff), lv.PART_MAIN)
oflabel.align(lv.ALIGN_CENTER, 0, -45)
var iflabel = lv.label(scr)
if f20 != nil iflabel.set_style_text_font(f20, lv.PART_MAIN | lv.STATE_DEFAULT) end
iflabel.set_text(string.format(" Inflow = %.2f°C", data['DS18B20-2']['Temperature']))
iflabel.set_style_text_color(lv.color_hex(0xffffff), lv.PART_MAIN)
iflabel.align(lv.ALIGN_CENTER, 0, -5)
end

tasmota.add_cron("*/5 * * * * *", f, "every_5_seconds")

#- display wifi strength indicator icon (for professionals ;) -#
wifi_icon = lv_wifi_arcs_icon(stat_line)    # the widget takes care of positioning and driver stuff
clock_icon = lv_clock_icon(stat_line)
```
And this is what I see on the display after a reboot. The text becomes fuzzier as each 5 seconds go by...

[image]

nicandris, 2024-03-02:
Hey @richardsg307 not sure if you ever figured it out, the issue is that you are creating a new label every time in your method with `var oflabel = lv.label(scr)`. I used your code (thank you 🙇 ) but adapted it to get mqtt msgs and display them [here ](https://gist.github.com/nicandris/fe84da719278055c8d69156137bab0c1#file-autoexec-be)if you want to take a look

richardsg307, 2024-03-07:
> Hey @richardsg307 not sure if you ever figured it out, the issue is that you are creating a new label every time in your method with `var oflabel = lv.label(scr)`. I used your code (thank you 🙇 ) [...]

Thanks, I'll have a look.
I did overcome the problem but I did it by moving on to HASPmota which takes care of overwriting for you. I did a full write-up of my final project [here](https://github.com/arendst/Tasmota/discussions/20429)