Xinyuan-LilyGO/T-Display-S3 #23: CircuitPython support

tyeth, 2022-10-11:
Hey, having read the requesting PID issue on adafruit, https://github.com/adafruit/circuitpython/pull/6037#issuecomment-1085131782 , it seems someone got an email reply from raspberryPi, and forwarded it to two of the issue participants, so maybe the process can proceed? And then also for this board, the T-Display-S3, and the T-QT-Pro.

jeremyamoore, 2022-12-05:
Looks like a PID has been assigned for the T-Display S3 ... https://github.com/espressif/usb-pids/blob/main/allocated-pids.txt

0x813E | LILYGO T-Display-S3 - Arduino
0x813F | LILYGO T-Display-S3 - CircuitPython
0x8140 | LILYGO T-Display-S3 - UF2 Bootloader

tyeth, 2022-12-22:
Yeah sort of. I actually wanted the S3 support, and the T-Sim/T-PCI-E supported, which didn't have PIDs at the time, and there was a long running issue about assigning some officially that seemed stalled (the one for C3/rp2040) but was following the official process.
 One can't have circuitpython/micropython without the boards pid/vids defined, so having them officially agreed or at least mentioned in this repo would remove the duplicated effort.

Long story short the pid/vids are now defined and therefore circuitpython support can proceed.

tyeth, 2023-02-27:
It's probably no use to anyone yet, but got the board playing in circuitpython (pinouts defined), but no display working yet. 
Might save someone duplicated effort: https://github.com/tyeth/circuitpython/tree/add-lilygo-T-Display-S3

tyeth, 2024-02-01:
Subscribers / Readers will be no doubt pleased to hear that Scott (lead developer of CircuitPython) has added / re-enabled parallel display support in this pull request, also including a T-DisplayS3 build:
https://github.com/adafruit/circuitpython/pull/8862

I've tested the UF2 from CI (github actions build artifacts) and it's seems to be working fine. 
I believe it should be released as part of CircuitPython v9.0.0-alpha.1

Let this be the best year of CircuitPython so far...

closed 2024-02-19 by tyeth