Xinyuan-LilyGO/T-Display-S3 #24: Incorrect pin number in schematic?

alandsidel, 2022-10-12:
The schematic shows the "boot" button connected to pin 32 / GPIO0 and the "Key" button connected to pin 11 / IO35 with a 100k pull-up.  The pin numbers don't correspond to the pin numbers on the ESP32-S3 so I don't know what they're in reference to.

GPIO0 does work for using the boot button, but GPIO35 does not seem to work for using the other button.

Is 35 the correct pin to use that button?  My PCBs are marked V1.2 on the silk screen.

LilyGO, 2022-10-13:
I didn't find the schematic to be questionable. Are you asking the wrong board. If you insist, please attach a picture.

closed 2022-10-13 by LilyGO

alandsidel, 2022-10-13:
> I didn't find the schematic to be questionable. Are you asking the wrong board. If you insist, please attach a picture.

What do you need a picture of?

LilyGO, 2022-10-14:
You think the wrong picture.