Xinyuan-LilyGO/T-Display-S3 #26: What connector to power by battery?

sulph68, 2022-10-13:
There are 2 physical connectors on this board. The JST and another connector near the reset button.
What are these connectors for?
i want to power the board from a Li-ion battery. Which connector should i use? and how can i read the battery voltage?

LilyGO, 2022-10-14:
2Pin is the battery connector. 4Pin is the IIC connector. It has the pin order on it (small font). Obtain the battery voltage by multiplying the IO4 analog value by 2.

closed 2022-10-14 by LilyGO

sulph68, 2022-10-14:
What’s the minimum voltage required to drive the board? I supplied a
3.7-4.2V single cell li-ion battery and the board doesn’t work. The green
led flashes once when I press reset, but the board doesn’t power up.