Xinyuan-LilyGO/T-Display-S3 #33: Turn off display?

nicpottier, 2022-10-25:
Any hints to turning off the display? Tried this and no luck:

```
  pinMode(15, OUTPUT);
  digitalWrite(15, LOW);
```

nicpottier, 2022-10-25:
```
// turn display off
tft.writecommand(ST7789_DISPOFF);

// turn display on
tft.writecommand(ST7789_DISPON);
```

closed 2022-10-25 by nicpottier

nicpottier, 2022-10-25:
Brightness control:

```
 ledcSetup(0, 10000, 8);
 ledcAttachPin(38, 0);

 // turn off backlight
 ledcWrite(0, 0);

 // turn on backlight
 ledcWrite(0, 50);
```

wichjan, 2023-06-21:
When switching off the display and backlight to save some battery power via:
```
ledcWrite(0, 0);  // turn off backlight
tft.writecommand(ST7789_DISPOFF); // turn TFT off
```
or even with an  additional
`digitalWrite(38, LOW);`
does NOT result in a complete shutdown of the display – although everything is dark – as can be seen when measuring the current consumption: the board is drawing approx. 30mA more current compared to the consumtion when the board is running with the display never being activated in `setup()`.

Does anyone have a suggestion how the really switch of all related components of the display?