Xinyuan-LilyGO/T-Display-S3 #35: Example TFT results in blank screen

roblatour, 2022-10-26:
The example found here:
https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/example/tft

compiles, but when run results in a blank screen - no output is displayed on the screen.

Info is written to the output debug log, but nothing to the screen.

The example at:
https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/example/factory
works fine.

LilyGO, 2022-10-28:
Please use the tft library provided by us. There will be changes here.

roblatour, 2022-10-28:
I did use the tft library provided by you - and as noted above the other example (factory) works fine.

However, I raised this issue because the tft example did not work for me.  

As far as I can see there have not been any coding changes to resolve this, so I'm not sure why this issue is closed?

LilyGO, 2022-10-31:
Are you using the Arduino IDE (Arduino 2.0.0). This version will automatically update the library version. This will overwrite the files you have dragged into the \Arduino\libraries directory. So cause your use of tft routines will fail. You can switch to Arduino 1.8.1x version and it will be solved.

roblatour, 2022-11-01:
actually, having your libraries where they are 
...  Arduino\libraries\TFT_eSPI
is working fine for the factory example as well as another program I am writing.
Have you tried the tft example lately?  Is it working for you?

LilyGO, 2022-11-02:
Everything works fine for me with this library, you can try it with Arduino version 1.8.19 (re-override the TFT library)

roblatour, 2022-11-03:
yes, its working now with arduino 2.0.1 ; not sure what the problem was but I'm sure it was on my end - sorry about that.

closed 2022-11-03 by roblatour