Xinyuan-LilyGO/T-Display-S3 #36: Do you have 3D file of board with screen attached?

t-rekttt, 2022-10-26:
I downloaded and opened your 3D file but there was only the ESP32S3 board and no screen, hence it's difficult for me to design a case for it. It would be great if you can provide 3D file with screen included.

tyeth, 2022-12-11:
@mmMicky is it possible to get both the screen versions as 3d files. The rectangular non-touch version, and the newer touch version with rounded glass. Thanks!

mmMicky, 2022-12-12:
https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/3D_file/glass_screen.stp

I uploaded the screen file. actually using the need to combine them together

t-rekttt, 2022-12-12:
> https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/3D_file/glass_screen.stp
> 
> I uploaded the screen file. actually using the need to combine them together

I'm having this problem when trying to import the file into Autodesk Fusion 360. Please check.
[image]

mmMicky, 2022-12-12:
I use Rhino 7 to open fine.

tyeth, 2022-12-12:
Thanks @mmMickey, that's brilliant.
@t-rekttt thanks for the warning, it opens with freecad (file->import)
[image]

tyeth, 2022-12-13:
[TDisplayS3touch_stl_fbx_step.zip](https://github.com/Xinyuan-LilyGO/T-Display-S3/files/10220939/TDisplayS3touch_stl_fbx_step.zip)
Added version together, it's approximately positioned 1.5mm from top edge and 5.5mm from board to glass surround which matches roughly the unit in my hands, and matches the normal t-displays3 if you measure to the top of LCD.
[image]

closed 2023-01-06 by mmMicky