Xinyuan-LilyGO/T-Display-S3 #37: The factory example is noct working

wecapp, 2022-10-31:
The factory example doesn't work. Is there a correct version?

bratoff, 2022-10-31:
It worked for me.  Did you follow all the instructions exactly?  What error message are you getting?

LilyGO, 2022-11-01:
Be careful not to upgrade the TFT library we provide when using the Arduino IDE

oldbug62, 2022-11-01:
I have the same isue, the display just blinkin what can i do?

wecapp, 2022-11-01:
I found that a word in factory.cpp appears to be misspelled. Instead of subscribe, the skecht says subsribe. Isn't that a mistake?

LilyGO, 2022-11-02:
The factory example will use all the libraries in lib, please drag and drop to the C:\Users\NAME\Documents\Arduino\libraries directory

bratoff, 2022-11-02:
A better place to put the libraries would be C:\Users\NAME\AppData\Local\Arduino15\packages\esp32\hardware\esp32\VERSION\libraries
That way they won't interfere with the Arduino-specific libraries or any other cores you have installed.

nfcrugby, 2022-11-08:
P.s. Am using Ubuntu 22.04.

Vn-ChemGio, 2022-11-14:
I has the same issue in Windows 10 and Arduino IDE 2.0.2 when flash file factory.ino

Vn-ChemGio, 2022-11-15:
> The factory example doesn't work. Is there a correct version?

I've see the reason
root cause by we didn't complete the setup

In my case, I forgot complete step : **Select the settings as shown.** in document
[image]
--
Note: You can also change timezone, wifi SSID and password in pin_config.h to use your wifi

nfcrugby, 2022-11-15:
Hi Windblade
I had most of these selected.
Can you just confirm "Core Debug Level:" in English.
I also have Programmer selected as Esptool under Tools and cannot turn that selection off for some reason! 
Any help there would be apreciated

Can I ask you exactly which Factory Version sketch you used? Copy of the link on GitHub would be best.
Thanks

Vn-ChemGio, 2022-11-15:
Core Debug Level

> Hi Windblade I had most of these selected. Can you just confirm "Core Debug Level:" in English. I also have Programmer selected as Esptool under Tools and cannot turn that selection off for some [...]

Core Debug Level can be choise is None
Make sure you choose exactly Flash Size, Flash Mode in image above
--
The code in master brand on this repository

---
You can ping me in Telegram and give me your teamview ID for check

nfcrugby, 2022-11-15:
Hi Again,
Thanks for this advice Vn-ChemGio which I have implemented.

After downloading the complete repo and renaming the folder to remove the dashes (---) I unzipped and manually put everything into the Arduino Library folder. Then I loaded the factory code onto Arduino IDE. BUT --  when I tried to compile I got the error message:
OneButton.h: no such file or directory.

I checked and it is there as are other header files.
I then read somewhere that the header files have to be moved out of the Library folder but my attempts at this have so far failed.  It is not easy for me, a relative newbie. Can anybody give me more precise instructions or tell me where to look?

(I did not include the full error message which is about 10 pages long and beyond my understanding.)
Again, for info, I am working on Ubuntu 22.04.

Many Thanks for your patience.

Vn-ChemGio, 2022-11-16:
Make sure you have installed all library bellow before compiler 
- ArduinoTEA5767
- OneButton
- Time
- Timezone
- TouchLib

csylvain, 2022-11-17:
i discovered that 
1. after copying all the directories in 'lib' in this repo to ~/Arduino/libraries, then
2. move all files from TouchLib/src to TouchLib, 
3. and remove 'src' from TouchLib. 

Verify/Compile will then work without generating any error messages.

linjian128, 2022-11-19:
@csylvain     Hi,   I still cannot  succeed to compile  "lv_demos", seems the "demo" folder is not included in the compile path. 
Did you have some other configuration after the readme?

mmMicky, 2022-11-21:
You can try to use platformIO to compile. I have specified the library directory and demo compilation path on this tool. Just open the example independently and click upload.

mmMicky, 2022-12-02:
Problem seems to be resolved. I will close this issue

closed 2022-12-02 by mmMicky