Xinyuan-LilyGO/T-Display-S3 #38: Usb port disappears

SoniaZotz, 2022-11-01:
on reset of the board  - USB Serial Port Com4 shows up 

couple second-later it disappears 

cannot load the  ESP

SoniaZotz, 2022-11-01:
it is running ...  **How do I stop it ??**

/* Hello World Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"

void app_main(void)
{
    printf("Hello world!\n");

    /* Print chip information */
    esp_chip_info_t chip_info;
    esp_chip_info(&chip_info);
    printf("This is %s chip with %d CPU core(s), WiFi%s%s, ",
            CONFIG_IDF_TARGET,
            chip_info.cores,
            (chip_info.features & CHIP_FEATURE_BT) ? "/BT" : "",
            (chip_info.features & CHIP_FEATURE_BLE) ? "/BLE" : "");

    printf("silicon revision %d, ", chip_info.revision);

    printf("%dMB %s flash\n", spi_flash_get_chip_size() / (1024 * 1024),
            (chip_info.features & CHIP_FEATURE_EMB_FLASH) ? "embedded" : "external");

    printf("Minimum free heap size: %d bytes\n", esp_get_minimum_free_heap_size());

    for (int i = 10; i >= 0; i--) {
        printf("Restarting in %d seconds...\n", i);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
    printf("Restarting now.\n");
    fflush(stdout);
    esp_restart();
}

LilyGO, 2022-11-01:
USB CDC is also a task for ESP32. It restarts or resets to re-enumerate.

SoniaZotz, 2022-11-01:
but I'm in a lockup system - cannot access the USB port because the system is constantly **rebooting**  as shown in the program above  

how do I **break out of this program** if my USB is constantly being reset USB CDC is also a task for ESP32. as you say ...**.it restarts or resets to re-enumerate.**

SoniaZotz, 2022-11-01:
[image]
[image]

LilyGO, 2022-11-02:
Press and hold the BOOT button, then short press the reset button, and then release the BOOT button to enter the BOOTLOADER mode.

SoniaZotz, 2022-11-02:
**### Thank you**

[image]

closed 2022-11-07 by liangyingy