Xinyuan-LilyGO/T-Display-S3 #39: documentation on external antenna

roblatour, 2022-11-01:
Missing documentation on external antenna.

Is it ok to use the t-display without an external antenna attached?

Other than plugging in an external antenna do you have to do anything to use it?  For example, setting a jumper?

LilyGO, 2022-11-02:
It can use the onboard antenna. The jumper is only necessary when you need to use an external antenna.

roblatour, 2022-11-03:
where is the jumper on the board? ( a photo would be helpful )

achtnullzwei, 2022-11-06:
I am interested as well. Wouldnbe great if it can be documented how to set the bosrd up for external antenna usage.

mmMicky, 2022-11-07:
[image]
Switching to external requires rotating this resistor 90 degrees.

closed 2022-11-07 by liangyingy