Xinyuan-LilyGO/T-Display-S3 #40: Voltage and power of signal outputs

SimLV, 2022-11-04:
Could you please put which voltage it could send on signal outputs and which load is acceptable on main page?

mmMicky, 2022-11-07:
I don't understand you

SimLV, 2022-11-10:
Excuse me.
I mean how much voltage I may drain from signal outputs. 
How much voltage I may put into signal inputs (are they 5V or 3.3V?).
and how many milliamps without risk of damage to the board.

I.e. from SoC datasheet
[image]

[image]

mmMicky, 2022-11-10:
The circuit signal needs a 3.3V voltage, which can ensure that the circuit board will not be burned and work normally.

closed 2023-01-06 by mmMicky