Xinyuan-LilyGO/T-Display-S3 #42: Can not charge the battery via IO PIN

fjulian79, 2022-11-07:
Hi .. 

As the battery charger is connected to VBUS, but VBUS is not connected to a IO pin, it is not possible to use other power sources than USB. I would like to use a solar panel to charge the battery!

br J.

mmMicky, 2022-11-12:
We have not considered this situation. Maybe you can solve it by soldering a wire from VBUS.

fjulian79, 2022-11-12:
Of course I can do this as workaround. But it would be a improvement in a upcoming revision to solve this issue.

tyeth, 2022-11-12:
While your improving the battery situation, can you start using a charging chip that supports the much safer LiFePO4 batteries, lithium iron phosphate, with a bridged solder pad to select type unless is automatic by chip. It would help a lot and I believe the voltage range is fine through the life of the charge, thank you.

mmMicky, 2022-12-02:
Problem seems to be resolved. I will close this issue

closed 2022-12-02 by mmMicky

fjulian79, 2022-12-02:
And will there be an improvement in future?

mmMicky, 2022-12-03:
> And will there be an improvement in future?

I'm not sure if there will be changes in the future because of this requirement. But if there are a large number of customers who need this feature. I think it will be re-added this functionality.

DevJulian1966, 2023-11-01:
I've just came across this as I tried to power the board via 5V pin and GND directly. The charging voltage jumps up to 5V, because of backpowering.
If you're using an unprotected battery it might blow up in this case! 
Any ideas how to charge the battery via 5V pin? I specifically want to use a wireless charger on this pin without USB.

teastainGit, 2023-11-01:
LilyGO made a small mistake with the LilyGO T-Display S3, in that they underestimated how many people would want to power the board from a 5V pin and not the convenient USB port! (VBUS)
You must only use the USB for 5V, as VBUS is not brought out to the edge pins.
May I ask why not uses USB?
For what it is worth they changed it dramatically on the LilyGO T-Display S3 AMOLED !!!
-Terry

DevJulian1966, 2023-11-01:
Hi Terry, thanks for the quick supply!
I want to use the board in a remote, which get's charged wirelessly when put back on it's base.
Any other option than using an USB C break out board / female connector or removing the delicate screen to solder a wire to VBUS?
-Julian

teastainGit, 2023-11-01:
Hi, Julian.
Why not use the provided USB? I know that somehow this is not the answer that you are looking for but the wireless charger pad could just plug into the existing USB, no?
Sorry I can't help!
-Terry