Xinyuan-LilyGO/T-Display-S3 #44: T-Display S3 bricked red + green LED only

68artik, 2022-11-09:
Hi!
The only I get is the red and green LED. I cannot boot it at all. Seems like it needs total firmware upload. I'm on the Mac M1.
Please advise.

mmMicky, 2022-11-10:
Do you mean using mac can not identify the port?

teastainGit, 2022-11-13:
68artik Mine seemed bricked as well. I spent a week getting it running. I put all my thoughts into a GitHub repo here:  https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples
if you can compile and download your sketch but no LCD image or no activity on the Serial Monitor:
1) The LCD must be enabled by outputting GPIO 15 high
2) The Serial Monitor is not enabled by default
Hope this helps! -Terry

kjarbour, 2022-11-20:
I just received the same display and through ide placed some code on it from a Volo project and now it's gone dark, green and red led's on. Any luck with your display?

teastainGit, 2022-11-20:
> Hi, it's me again from Facebook. Mine is running fine and I created a repo on GitHub:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples
If you want to get it running this will help!

mmMicky, 2022-12-02:
Problem seems to be resolved. I will close this issue

closed 2022-12-02 by mmMicky