Xinyuan-LilyGO/T-Display-S3 #45: T-Display S3 lipo do not power display

redPanther, 2022-11-14:
Hi,
when board is powered via USB all things are fine (display and so on), but when powerd via LiPo (1.1Ah) display won't work (Voltage on 3V pin is fine).

Any suggestions?

teastainGit, 2022-11-15:
Hi, you could check out my Repo here:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples
LilyGO-T-display-S3-setup.txt

Solutions to common startup issues:
C) Can run on USB power but not battery connector:
	Add #define LCDpin 15		// just before the setup and then in setup add:
	pinMode(LCDpin, OUTPUT); 	//"triggers the LCD backlight,
	digitalWrite(LCDpin, HIGH);)	//on startup"
	
	(LilyGO variously refers to them as "LCD power on" OR "PIN_POWER_ON")
	Thanks to user RobLatour for finding this cure for the battery issue!
-Terry

redPanther, 2022-11-15:
thx thats worked!

I would like to see some more curcial infos on this image: https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/image/T-DISPLAY-S3.jpg

LCD_BACKLIGHT_PIN GPIO 15
BOOT GPIO 0
Connector with pin 43/44 is UART0
USB connector is usb OTG (and not usb serial as other esp32 boards)
also the external antenna should be advertised (is it simply usable or must be magic done to use it?)

Also "LCD_BAT_VOLT 04" is hard to understand. should be something like "BAT_ADC 04" like in the schematics.  Also pointing to the battery connector is not helpful. In the place of "LCD_BAT_VOLT 04" should be something like "4.2V LiPo battery". Next to it the BAT_ADC pin could be mentioned.

... just my ideas, perhaps there is some useful stuff in it. 
My issue is resolved, so I close this issue. thx again for help!

closed 2022-11-15 by redPanther