Xinyuan-LilyGO/T-Display-S3 #48: "Tips" section of READMe is not understandable

teastainGit, 2022-11-16:
Hi! I'm sorry to say this, but as an experienced Arduino maker, the "Tips" section of READMe is not understandable.
Firstly when the board is first plugged in there is a grey screen saying connection to Xinyuan... ... ...
what does this do?
Second the Tips are more confusing than helpful.
Once set up, the board behaves like a normal Arduino or ES32!
I really like the product but the new users deserve more support.
(Also I started my own GitHub repo for new users of T-Display S3 here:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples)

Everyone is free to help improve this resource!
-Terry

bratoff, 2022-11-17:
My bigger beef is that the directions have you (potentially) replace several standard Arduino libraries with the particular version Lilygo's developer used, which in many cases is much older than the current version.  In most cases, the Tips should just have you make sure the latest versions are installed, using the Arduino Library Manager.

The unpleasant exception I have found is that there have been nonstandard modifications made to the TFT_eSPI library, which is also used by many other Lilygo boards.  All you should ever have to do to configure TFT_eSPI  is uncomment the correct setup file in User_Setup_Select.h, or possibly create a new Setupnnn_xxxxxx.h file in the User_Setups folder and add a line for it in User_Setup.h if a suitable setup does not already exist.  This technique works for just about every other Lilygo board, but there appear to have been additional code changes made to the TFT_eSPI library in the T-DisplayS3 repository.  This is unacceptable, as it creates conflicts when you're developing for multiple board types and makes it impossible to stay current with the latest official copy of the library.  If these changes are really necessary and not just the result of a lack of understanding, Lilygo should make them on the latest official version of TFT_eSPI and then submit them as a PR to the library's maintainer.

mmMicky, 2022-11-18:
> My bigger beef is that the directions have you (potentially) replace several standard Arduino libraries with the particular version Lilygo's developer used, which in many cases is much older than [...]

The I8080 interface of the TFT_eSPI library to the ESP32S3 chip is driven by software simulation. The library itself uses register operations for speed. In fact, the GPIO register width of ESP32S3 is only 32 bits, and one register is not enough to allocate more than 40 pins. So the gpio control register of ESP32S3 has two groups. In TFT_eSPI, it only takes effect on the low pin registers. So I had to modify the driver inside the library to support high pins.

mmMicky, 2022-11-18:
At the same time, I will place the modified places in the TFT_eSPI directory in the form of patches. You can use this patch to supplement the new version TFT_eSPI.

mmMicky, 2022-11-18:
https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/lib/TFT_eSPI/micky_commit.patch

bratoff, 2022-11-18:
> https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/lib/TFT_eSPI/micky_commit.patch

Thank you - I will apply the patches to the latest official TFT_eSPI and see how that works.

If it checks out, why not submit it as a pull request to the author of TFT_eSPI?

bratoff, 2022-11-21:
> > https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/lib/TFT_eSPI/micky_commit.patch
> 
> Thank you - I will apply the patches to the latest official TFT_eSPI and see how that works.
> 
> If it checks out, why not submit it as a pull request to the author of TFT_eSPI?

I have successfully merged Micky's patch with the latest official TFT_eSPI (version 2.4.79).  Several differences from Micky's patch were necessary:
- ST7789_init.h should NOT be modified, as this breaks any user config where TFT_INVERSION_ON was not explicitly specified
- in TFT_eSPI_ESP32_S3, the name of the macro for the clear mask had to be changed to be consistent with changes the library owner had made
- I created a new user setup file (User_Setup151_ST7789_TDisplay_S3.h) specifically for this board so that the patch would not break any other boards that were using User_Setups/Setup24_ST7789.h
- Because of a change in the library owner's code, #define TFT_RGB_ORDER TFT_BGR must be added to the user config file.  This also proved necessary for the original T-Display and the T-Display_RP2040 boards

Here is a zip of my final patch.  It is no longer necessary to completely overwrite the official library in your libraries folder.  Instead, these files can just be copied on top of the official version 2.4.79 with no other changes necessary:

[TFT_eSPI - T-DisplayS3 patch.zip](https://github.com/Xinyuan-LilyGO/T-Display-S3/files/10060751/TFT_eSPI.-.T-DisplayS3.patch.zip)

teastainGit, 2022-11-22:
Great!
Man, I could not even...

Cheers !
Terry Stainton

(519) 729-1047  (Call or text)
***@***.***

dapug, 2022-12-23:
Shouldn't we work with the TFT_eSPI author to make changes for this board?  It would not affect other existing users of TFT_eSPI, right?  It only takes effect with TFT_DATA_PIN_OFFSET_EN build_flags, correct?

The problem I am having is, I have a project using source I didnt write, and it dynamically is pulling the latest TFT_eSPI from the web, and I cannot seem to "patch" anything locally to take effect.  Even if I could... this patch should be in the official lib, otherwise, to use t-display-s3 is very "hacky".

bratoff, 2022-12-27:
> Shouldn't we work with the TFT_eSPI author to make changes for this board? It would not affect other existing users of TFT_eSPI, right? It only takes effect with TFT_DATA_PIN_OFFSET_EN build_flags, [...]

I agree.  As Micky is the author of the patch and I jsut cleaned up a few things to make it work with the latest released TFT_eSPI, I was hoping Micky would make a pull request on the TFT_eSPI project, which would enable the TFT_eSPI author to merge the patch into a future release.

ElectricDream5, 2023-01-02:
Thank you Bratoff for posting this fix. I was pulling my hair for days after updating TFT_ESPI to version 4.79 on IDE 2.03. The screen is working once again.

I agree with OP TeaStainGit regarding the tips section needing improvements. They have added useful information in the past couple of months, but beginners like me could use a bit more hand holding during set up.Of course, this has to do with the TFT_ESP Lib adaptation for this board as well.

mmMicky, 2023-01-04:
I have submitted a pull request to the TFT_eSPI author.
https://github.com/Bodmer/TFT_eSPI/pull/2296

ElectricDream5, 2023-01-05:
Thank you for all the work you've done to make these great little boards more accessible.

teastainGit, 2023-01-05:
Hi Electric Dream,
These issue boards are confusing!
Who are you referring to with your thanks?
I just need to know if my repository is helpful.

Cheers !
Terry

ElectricDream5, 2023-01-05:
Hi Terry,

At that point in the thread I was responding to Mickey and thanking him/her for the work to make it easier to start up with the T Display S3. Of course, your point still stands, the instructions can be improved, so thanks for pointing that out. Well worth the effort.

Andres

Bodmer, 2023-01-08:
The patch pull request has been applied to this TFT_eSPI master branch:

https://github.com/Bodmer/TFT_eSPI/tree/T-Display-S3

Please test before I merge into the master branch as I do not have one of these boards.

dapug, 2023-01-08:
> I have submitted a pull request to the TFT_eSPI author. [Bodmer/TFT_eSPI#2296](https://github.com/Bodmer/TFT_eSPI/pull/2296)

Did that patch include updates that @bratoff made?  And to @Bodmer 's question about testing... we can test if it works with the T-Display-S3 board, but how can we be sure other boards don't break?

mmMicky, 2023-01-09:
> The patch pull request has been applied to this TFT_eSPI master branch:
> 
> https://github.com/Bodmer/TFT_eSPI/tree/T-Display-S3
> 
> Please test before I merge into the master branch as I do not have one of these boards.

I used the library in this link with the alphaBlend_Test example and the example in the link below
carry out testing.

https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/example/tft/tft.ino#L21

works fine.

I suggest adding the following initialization commands to this model and using macros to control whether to enable or disable, sometimes it will make the screen display better.

https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/f155cb3835dd8c9fadf6a65f4cbd613e26d881f9/example/tft/tft.ino#L21

Bodmer, 2023-01-13:
The master library of TFT_eSPI has been updated to include support for 8 bit parallel bus using high (>31) GPIO numbers.

All display data bits (D0-7) must use either high (>=32) or low (<=31) processor GPIO numbers since a single register write is used to output all data bits at the same time.

svdrummer, 2023-01-15:
Other than uncommenting line  131
 #include <User_Setups/Setup206_LilyGo_T_Display_S3.h>

is there anything else. The text is all over the pace in a T3

I downloaded the zip file from github as the Arduino version is old.

owiecc, 2023-02-10:
I tried to run with TFT_ESPI v2.50 but I get a garbled or offset display. I uncommented `#include <User_Setups/Setup206_LilyGo_T_Display_S3.h>` in `User_Setup_Select.h` @ElectricDream5 did you do anything else?

Bodmer, 2023-02-10:
Did you comment out the default setup line like this:
//#include <User_Setup.h>

owiecc, 2023-02-10:
That helped, thanks.

closed 2023-03-13 by lewisxhe