Xinyuan-LilyGO/T-Display-S3 #49: failed to compile  example

linjian128, 2022-11-19:
Hi, 

I have done all settings in readme, and trying to upload an example to the board.   

But I was failed to compile the example in Arduino IDE(2.0.1),  getting the message 
  "...    example\lv_demos\lv_demos.ino:3:10: fatal error: TouchLib.h: No such file or directory  "

I'm sure  I have copeid all files in lib to the Arduino libraries folder,   is there any suggestion? thanks.

Vn-ChemGio, 2022-11-19:
@linjian128  your issue can be resolved same in this page 
https://github.com/Xinyuan-LilyGO/T-Display-S3/issues/37

linjian128, 2022-11-19:
Thanks, I also find the root cause should be the wrong layout of the library "Touchlib"

It should be:
      Arduino
          |-libraries
              |- TouchLib
                  |- TouchLib.h

rather than: 
      Arduino
          |-libraries
              |- TouchLib
                  | - src
                      |- TouchLib.h

closed 2022-11-19 by linjian128