Xinyuan-LilyGO/T-Display-S3 #51: Powering device from external source

OXOTHuK1976, 2022-11-20:
How to power board from external 5v source? Tried  battery connector but no succes.  It only powering from type-c

mmMicky, 2022-11-21:
[image]
From here the port can be powered with 5V.

closed 2022-11-21 by mmMicky

OXOTHuK1976, 2022-11-21:
Thats not work,  external 5v does not power device from this pin. Get 0.05A  but not started.   So, it is broken board?

mmMicky, 2022-11-21:
I used the 5V and GND on the Raspberry Pi to connect to the board and it can be successfully lit.

OXOTHuK1976, 2022-11-21:
Ok, seems my board is broken,  I used 5v from laboratory power source and from usb, nothing happen. 
Thank you and good luck.

mmMicky, 2022-11-21:
[image]
Not necessarily damage. There is a PWR_EN label here on the schematic. It is controlled by either USB or GPIO. You need to make sure you have this pin high.

OXOTHuK1976, 2022-11-21:
Ok, thank you.  Im not strong in schematic.  Which pin I need pull up?  Or I need find this diode on board and pullup here?

mmMicky, 2022-11-21:
This label corresponds to GPIO15. You need to set this to high.

OXOTHuK1976, 2022-11-21:
Can`t find this pin on board. On your image it is LCD_POWER_ON,   but can`t find it physically.

mmMicky, 2022-11-21:
It is not drawn externally. Only inside the board.

OXOTHuK1976, 2022-11-21:
Thank you for your help. 
Finally crack the screen.  So question is not actual any more :)))
PS:  To everyone - be careful. The screen is very fragile, it cracked when I tried to insert the device into the breadboard.

wiegleyj, 2022-12-20:
I wish this was better documented.

For those that don't know. The pull-up/down state of GPIO pins is persistent when power is removed.

So, if you do:

```
  //For esp-idf programming
  #define PWR_EN 15
  gpio_reset_pin(PWR_EN);
```
and run that code sequence once while powered from the USB cable. This will do the trick permanently as the esp32 maintains its pullup state when power is removed. the gpio_reset_pin disconnects the GPIO from both input and output but does enable the pullup resistor on it (from docs: select gpio function, enable pullup and disable input and output).

Not sure of the code sequence on arduino programming for esp32. But I think `pinMode(15, INPUT_PULLUP);` will do. Otherwise you'll need to go through the `gpio_config` functionality.

In any case, any thing you do to cause the ESP32 chip to enable its pullup resistor on GPIO15 will be persistent across power cycles unless you change/select a different pullup configuration for that pin.

doh38, 2023-03-10:
> quence once while powered from the USB cable. This will do the trick permanently as the esp32 maintains its pullup state when power is removed. the gpio_reset_pin disconnects the GPIO from both [...]

Thanks, it indeed works perfectly on arduino with the line you gave. one needs to put in in the setup(), but that's not a surprise :)

Now the 1000000$ question... what happens then when usb is powered and ext 5V too? also does this charge the battery? Asking before toasting my board :)