Xinyuan-LilyGO/T-Display-S3 #52: ESPHome support

OXOTHuK1976, 2022-11-20:
It is possible to support ESPHome with this device/display?

mmMicky, 2022-11-21:
Not sure if the screen supports it.

OXOTHuK1976, 2022-11-21:
Screen itself can be used in esphome with external library with your patch.   
example: https://github.com/landonr/lilygo-tdisplays3-esphome

but will be great to have native esphome support.

Jason2866, 2023-01-08:
Or use Tasmota ;-) Fully supported.

eisaheinig, 2023-01-25:
> Or use Tasmota ;-) Fully supported.

i was not able to find it on this list: https://templates.blakadder.com/esp32.html

Jason2866, 2023-01-25:
@eisaheinig Yeah, missing there. Needs adding...
Install from here https://tasmota.github.io/install/ using this build
[image]
and configure it via Tasmota Menu Autoconfig. There it is.

lewisxhe, 2023-03-13:
If you still have problems, please open it again

closed 2023-03-13 by lewisxhe