Xinyuan-LilyGO/T-Display-S3 #55: WAFER-100W-4A part 2

AlexanderSkvortsov, 2022-12-01:
How to find the PN of the wired connector for the connector WAFER-100W-4A ?

mmMicky, 2022-12-02:
I don't know what this connector is.

teastainGit, 2022-12-03:
It is on the supplied schematic!

[image]

sthaugen, 2022-12-03:
i'm using this cable:
SH1.0 JST 4p.   
https://www.aliexpress.com/item/1005001569197247.html

mmMicky, 2022-12-03:
Well, I'm just dealing with software issues. There is no emphasis on the details of the hardware circuit.

bratoff, 2022-12-03:
I'm using these, although I did have to rearrange the wires so that the red and black were on the 3V3 and GND pins.

https://smile.amazon.com/dp/B08RHGT3W3?psc=1&ref=ppx_yo2ov_dt_b_product_details

tyeth, 2022-12-03:
It's the qwiic / stemmaQT / JST-SH connector. I'm using these at 80pence each with various breakout boards (adafruit / sparkfun / laskakit.cz ). Cheap enough to cut and solder to boards without qwiic sockets.
 £0.80 per cable 10/15/20cm, £1.30 for 30cm and £1.50 for 40cm.
https://thepihut.com/search?q=qwiic&narrow_by=&sort_by=price_min_to_max&page=1

teastainGit, 2022-12-03:
Hi, I have mostly M5Stack sensors with 2.5 pitch Grove connectors, so I made a converter!
Cheers !
Terry

AlexanderSkvortsov, 2022-12-04:
Thanks a lot of help. I have bought the qwiic connector, but will get in after 2-3 weeks only.  But why this connector is not include in the box the same as the JST power connector? It would save a lot of time.

tyeth, 2022-12-04:
I don't think it's standard enough _yet_, as most daughter boards (sensor breakout boards etc) don't have the matching socket. It would be more common to include dupont jumper cables like an arduino starter kit. 

Interestingly over the last year there has been a dramatic push by Adafruit to revamp all their boards with them, as the no-soldering route seems to enable access to a lot more users. Glad the connectors on there so thanks "LilyGo", but 75% of my sensors didn't come with the connector. I am now buying sensors with JST-SH/stemmaQT if possible, and adding cables to those that don't as the plug-and-play way is a winner.

closed 2023-01-06 by mmMicky