Xinyuan-LilyGO/T-Display-S3 #57: Please update Schematic !

teastainGit, 2022-12-02:
Hi mickey!
I am using two T-Display S3 units and supporting other users on LilyGO community.
In trying to help others I refer to the schematic but it does not seem complete or at least, up-to-date.
Example there 2 LEDs shown, but both are marked RED, but they are Red and Green.
There is a greyed-out WS2812 drawn in the middle of the current schematic. What is this?
Could we get a better Schematic?
Thanks, Terry

svdrummer, 2022-12-02:
If the AOT values for the battery could be added as well. Quite a few resistor values are missing. 
On youtube, the claim is 800Ma for the T-displayS3, which is quite high. Youtube shows removing and reducing the value.
Terry. (Another Terry)

mmMicky, 2022-12-02:
> 

Thank you for your support to lilygo. The gray WS2812 in the schematic does not exist in reality. There are two LED lights on the board, one is for charging indication (it will flicker when the battery is not connected, it will be always on when it is connected to charging, and it will be off in other states). The other one is connected to the same line as the backlight to be controlled together with the backlight.

mmMicky, 2022-12-02:
> If the AOT values for the battery could be added as well. Quite a few resistor values are missing. On youtube, the claim is 800Ma for the T-displayS3, which is quite high. Youtube shows removing and reducing the value. Terry. (Another Terry)

Sorry, I don't understand what parameter is battery AOT.

svdrummer, 2022-12-02:
Battery AOT(Adjust On Test) means some versions have one value resister and other version have another. I see the schematic is generic for several modules. Hence the WS2812 was for an RGB L.E.D.  As an engineer, I know documentation is behind development, however, product stability comes from documentation, hence why Adafruit does so well.

mmMicky, 2022-12-03:
There is no WS2812 led light on this board. I also don't understand why hardware engineers prevent it in this position.

mmMicky, 2022-12-03:
There is an independent io to obtain the sampling of the battery in the circuit board. But I think the current of the sampling resistor is too small. lead to inaccurate sampling. I think it may be more accurate if this resistor is replaced with a smaller value.

closed 2023-01-06 by mmMicky