Xinyuan-LilyGO/T-Display-S3 #58: are there I2C pins for this board?

contractorwolf, 2022-12-03:


tyeth, 2022-12-03:
[image]
43+44 as attached to the quIIC connector
The [qwiic ](https://www.sparkfun.com/qwiic) connector, also referred to as stemma QT or JST-SH 1.25mm, will give you easiest use of i2c, but the pins are broken out too.
You can always check the schematics for such things, but also the pinout diagram for other lilygo boards is more informative, so fair question

teastainGit, 2022-12-04:
mmicky, please close. It looks like OP bailed on his question, or it was answered by tyeth

contractorwolf, 2022-12-04:
Ok I see it now, thanks for the advice.  might want to add the standard SDA and SCL markings to the pin diagram.  I didnt know CLK_OUT and CLK_OUT2 would indicate I2C.  Thanks!

tyeth, 2022-12-04:
To be fair the pinout diagram is missing some of the more useful extra pin function labels, like I2C, which is why the question was asked. Maybe it can be updated to be better, more like the T-Sim7000G which at least highlights i2c
[image]

mmMicky, 2022-12-05:
ESP32 32 can map any peripheral to any gpio (except USB). I will close this question.

closed 2022-12-05 by mmMicky

tyeth, 2022-12-05:
I disagree, although the mux supports remapping, not every pin to every function. Make life easier for people and include helpful info, it's not about what can be done, but _what pins are the default_ for example if Wire.begin was called (and you have an IIC/I2C/qwiic connector for the purpose so it's defined although remappable).

 It's new users you're helping, veterans will waste their time in datasheets and framework sourcecode or already know about remuxing.

marshalleq, 2025-10-05:
I just came here due to the same problem with lilygo s3 v 1.2 lora device.  I'm still not sure what to connect to to be honest.  While you technically can connect to most pins, the software I've got is not configurable.  Some labels would sure be helpful.