Xinyuan-LilyGO/T-Display-S3 #59: Adding the lv_chart_example4 from lvgl examples to factory demo tile doesn't scroll properly.

tyeth, 2022-12-03:
The basic idea is to add the examples and demos into the build and call the 4th chart example on a new fourth tile, but the problem also happens if you just add the chart to the first tile (tv1 object). By "add the chart", I mean I'm duplicating chart4 from the examples but modifying the creation to be on one of the tiles, although the problem exists with just the unmodified ```lv_example_chart_4()``` call.

The symptom is the chart scrolls to the right after the user releases the touch. You cannot scroll the chart left and release because the chart scrolls right again after release.

tyeth, 2022-12-03:
I have tried adding the chart to a tile in the lv_demos.ino file in this repo, but the bug doesn't show and the chart can be scrolled successfully.

tyeth, 2022-12-03:
Partially Solved! f*cking touch interrupt. It's disabled in lv_demos but not in factory.
https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/example/factory/factory.ino#L3

 Disabling it stopped the bug, but now I want to know why, as there is only one conditional bit of code doing touch.read() so I don't get how it could be double firing or firing a second time late.

tyeth, 2022-12-03:
Although the graph does reliably scroll now, there have been side effects. Disabling the interrupt has caused other problems like the tooltips for chart points not showing, and the Touch Test label on tile3 doesn't reset back to Zero fingers.

mmMicky, 2022-12-08:
I guess there is a problem with the interrupt, that is, every time the last interrupt comes, I will get the coordinates of (0, 0) to read. Causes the icon to always move upwards.

mmMicky, 2022-12-08:
If the user experience is still not good, then the best function of interrupt is the function of waking up during sleep

mmMicky, 2022-12-12:
The effect of not using interrupts is no problem in my test. Regarding the battery test, I feel that the overall voltage is too high. It is possible that the current sampled by GPIO is too small to cause inaccuracy.

tyeth, 2023-01-02:
A similar issue on lvgl forum, points out the clean solution of not registering as pressed if number of fingers is zero.
https://forum.lvgl.io/t/slider-arc-roller-values-changes-to-zero-or-wrong-values-after-release/10587/2

I've patched mine with a renamed variable: https://github.com/Xinyuan-LilyGO/T-Display-S3/commit/2e5eb82e35853b495cbc55df37345e202ed5d1f8#diff-6eb686fe7cc6799a846fc7bcc08bf4d4566a82d71f003de9858308622b75d259L65-R84

lewisxhe, 2023-03-13:
If you still have problems, please restart

closed 2023-03-13 by lewisxhe