Xinyuan-LilyGO/T-Display-S3 #60: Micropython Driver and examples / documentation

Drew40B, 2022-12-03:
Hello,
I recently purchased an T-Display ESP32-S3R8 and I have flashed the firmware with micropython from here https://github.com/russhughes/st7789_mpy/tree/master/firmware/GENERIC_S3_7789

With using Thony I am able to connect to the board and work from the REPL. But I am not able to get anything to be displayed

Can you please update your documentation to include step by step instructions on how to setup micropython and how to get something to display?

teastainGit, 2022-12-04:
Hi, excerpt from my LilyGO T-Display S3 startup Repo:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples

Solutions to common startup issues:
A) Unit seems "bricked" 1
   LCD does not display, unit remains dark with no white backlight:
	Add #define LCDpin 15		// just before the setup and then in setup add:
	pinMode(LCDpin, OUTPUT); 	//"triggers the LCD backlight,
	digitalWrite(LCDpin, HIGH);)	//on startup"

Drew40B, 2022-12-04:
Thanks for the reply @teastainGit.

I noticed on your root page (https://github.com/Xinyuan-LilyGO/T-Display-S3) you have instructions for Arduino, platform IO and ESP-IDF. Since micropython is listed as a supported development language can you please add instructions for that as well?

I am new to ESP32 micropython  development. What I am really looking for is step by step instructions on how to properly flash the firmware for Micropython and then a simple working python example using the display.

iamfourh2e, 2022-12-07:
please do the micropython instruction. thanks in advance!

teastainGit, 2022-12-07:
OK, guys I think I stepped out of my lane here!
I responded with the info (In Arduino C language) with the instructions to turn on the backlight and enable battery power, etc.  
If you already have Python running,  but the LCD won't come on, please add Python commands to "turn on output 15" !!!
This board is very tricky to get running and program so that is why I made the teastainGIT Repo.
Cheers, Terry

mmMicky, 2022-12-08:
Documentation on micropython is supported here.
https://micropython.org/

The mpy firmware of T-Display-S3 can be compiled here.
https://github.com/Xinyuan-LilyGO/lilygo-micropython

Drew40B, 2022-12-08:
> Documentation on micropython is supported here. https://micropython.org/
> 
> The mpy firmware of T-Display-S3 can be compiled here. https://github.com/Xinyuan-LilyGO/lilygo-micropython

Would it be possible to add a link to these instructions to your readme?

lewisxhe, 2023-03-13:
If you still have problems, please restart

closed 2023-03-13 by lewisxhe