Xinyuan-LilyGO/T-Display-S3 #61: Battery issue

paryl, 2022-12-04:
When I first got the board, I plugged in USB and played with the demo. I then left it on the clock screen with only the battery connected, to test how long it would run that way. I have a standard 1100mAh lipo battery plugged into the board.

However, after flashing the board with new code, it shuts down when I disconnect USB, even though the battery is plugged in. I've tried a different (identical) battery, and it doesn't work either, though they work in other boards.

With only the battery plugged in, I read around 4V on the 5V pin, and 3.3 on the 3V pins, so the battery is definitely providing power.

I've looked through all of your code and can't find anything related to different behavior when on battery, so I don't think it's a software issue.

Any idea what could be happening?

teastainGit, 2022-12-04:
Hi, excerpt from my LilyGO T-Display S3 startup Repo:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples
Common problem:
C) Can run on USB power but not battery connector:
	Add #define LCDpin 15		// just before the setup and then in setup add:
	pinMode(LCDpin, OUTPUT); 	//"triggers the LCD backlight, and enables battery power
	digitalWrite(LCDpin, HIGH);)	//on startup"
	(LilyGO variously refers to them as "LCD power on" OR "PIN_POWER_ON")
	Thanks to user RobLatour for finding this cure for the battery issue!

paryl, 2022-12-04:
That solved it! Thank you so much!

closed 2022-12-04 by paryl