Xinyuan-LilyGO/T-Display-S3 #62: It seems like https://community.lilygo.cc/ is leaking

teastainGit, 2022-12-04:
Could The LilyGO company increase awareness of the user-supported website https://community.lilygo.cc/ ?
Many of the "Issues" in the Repo are Hardware and answered by fellow users !
There is also a Facebook group!

Cheers, Terry

tyeth, 2022-12-08:
I'm getting a problem with it saying no confirmation email yet so posts being moderated. There is no email and no way to resend.
Trying to change my account email and trigger something I get this error:

```
Please review the entered information:

Invalid login: 535 Error: authentication failed, system busy
```

teastainGit, 2022-12-08:
> I'm getting a problem with it saying no confirmation email yet so posts being moderated. There is no email and no way to resend. Trying to change my account email and trigger something I get this [...]

Hi tyeth, your community post:  "I have a 110mA max (16uS, but about ~63mA average)..." was online OK, so could you be more specific?

tyeth, 2022-12-08:
I saw the post appear, but it also said some moderation thing, and I saw that authentication error when I tried to do something.  I then logged out and back in again to see if the moderation message went away or I noticed anything different, I didnt.
I then tried to change my email to get a confirmation request, saw that error, and the change email option was there still when I logged out and back in again, so I cancelled the option to change my email.
I never got a confirmation request originally, nor when trying to change email and saw that error.
[image]

svdrummer, 2022-12-09:
If you send too many from one address, a lot of systems kick in the
DOS/anti spam filter.

lewisxhe, 2023-03-13:
LilyGo is a growing company. It is not doing well at present. It will get better later. Thank you for your suggestion