Xinyuan-LilyGO/T-Display-S3 #64: I2S enabling Arduino IDE

ArielIntel, 2022-12-10:
Is there an option to enable I2S in this board for using it with Arduino IDE ?

teastainGit, 2022-12-10:
Here is some light reading from Espressif :
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html

teastainGit, 2022-12-10:
> Is there an option to enable I2S in this board for using it with Arduino IDE ?

I2S is useable in Arduino IDE, yes!

It is a mash-up of I2C and SPI in my opinion and you could use any T-Display S3, GPIO except for GPIO0

ArielIntel, 2022-12-11:
Thanks. I found that article myself before but what it means? Is there a new driver? Which gpios can be used? Any practical examples?

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html

---------------------------------------------------------------------
Intel Technology Poland sp. z o.o.
ul. Slowackiego 173 | 80-298 Gdansk | Sad Rejonowy Gdansk Polnoc | VII Wydzial Gospodarczy Krajowego Rejestru Sadowego - KRS 101882 | NIP 957-07-52-316 | Kapital zakladowy 200.000 PLN.
Spolka oswiadcza, ze posiada status duzego przedsiebiorcy w rozumieniu ustawy z dnia 8 marca 2013 r. o przeciwdzialaniu nadmiernym opoznieniom w transakcjach handlowych.

Ta wiadomosc wraz z zalacznikami jest przeznaczona dla okreslonego adresata i moze zawierac informacje poufne. W razie przypadkowego otrzymania tej wiadomosci, prosimy o powiadomienie nadawcy oraz trwale jej usuniecie; jakiekolwiek przegladanie lub rozpowszechnianie jest zabronione.
This e-mail and any attachments may contain confidential material for the sole use of the intended recipient(s). If you are not the intended recipient, please contact the sender and delete all copies; any review or distribution by others is strictly prohibited.

ArielIntel, 2022-12-12:
teastainGi -  thanks for the answer but where can I configure the port which  I choose using?

ArielIntel, 2022-12-12:
> ### [Dec 10, 2022](#issuecomment-1345373186)

teastainGit- thanks for the answer but where can I configure the GPIO port which I choose using?

closed 2023-01-06 by mmMicky

contractorwolf, 2023-05-22:
@ArielIntel  what did you need to do? I am trying to use the T-Display S3 with an I2C device just with the qwiic connector and the display seems to not see the device at all.  Any advice would be appreciated.

teastainGit, 2023-05-22:
@contractorwolf The qwiic port works OK, what device are you trying to connect to?
Also: Are you using Arduino IDE?

contractorwolf, 2023-05-22:
@teastainGit I am trying to connect it to this keyboard via the qwiic connector:
https://www.solder.party/docs/bbq20kbd/

And I have used that keyboard as an I2C keyboard with a Seeeduino Xiao (using Arduino IDE)  without any issues.  But I dont know what to think here, doesnt seem  to see it. None of the Solder Party examples seem to work.  I cant tell what I maybe missing? Any advice?

teastainGit, 2023-05-22:
@contractorwolf With the newer ESP32s and especially the S3 I find that you need to replace Wire.begin();  with 
Wire.begin(43, 44);  //SDA, SCL which is the qwiic connector port!
Hope this helps!
-Terry

contractorwolf, 2023-05-23:
Yeah thanks @teastainGit, but that is what I am trying, I even tried reversing it just in case, but it is like an i2c device isnt registered on those pins for some reason.  I am really stumped.  Here is some of my code in case you can see something that I am missing:

`
#include "Arduino.h"
#include "TFT_eSPI.h" /* Please use the TFT library provided in the library. */
#include "pin_config.h"
#include <BBQ10Keyboard.h>
#include "Wire.h"

const String projectName = "BlackTerminal";

int current = 0;
int maxVal = 255;
int delayVal = 200;
int linePosition = 0;

long targetTimeout = 0;
int period = 500;

int currentPage = 0;
int charactersPerLine = 16;
int displayCharacters = 2 * charactersPerLine;

// special character int values
int charaterLeftArrow = 17;
int cursorChar = 18;
int characterUpArrow = 26;
int characterDownArrow = 27;

int commandKeyUp = 6;
int commandKeySelect = 7;
int commandKeyDelete = 8;
int commandKeyEnter = 10;
int commandKeyBack = 17;
int commandKeyDown = 18;

String outgoingMessage = "";
String incomingMessage = "";

BBQ10Keyboard keyboard;

TFT_eSPI tft = TFT_eSPI();

int xpos = 0;
int ypos = 0;
int yLine = 25;
int textSize = 4;

void setup()
{
    pinMode(PIN_POWER_ON, OUTPUT);
    digitalWrite(PIN_POWER_ON, HIGH);

    Serial.begin(9600);
    Serial.println("Hello T-Display-S3");

    tft.begin();
    tft.setRotation(3);
    tft.setSwapBytes(true);

    delay(2000);

    ledcSetup(0, 2000, 8);
    ledcAttachPin(PIN_LCD_BL, 0);
    ledcWrite(0, 120); // screen brightness 0-255

    Wire.begin(43, 44);  //SDA, SCL
    // start keyboard comms
    keyboard.begin();
    keyboard.setBacklight(0.5f);
[... 161 lines cut ...]

void deleteCharacterPressed()
{
  currentPage = 0;      

  outgoingMessage.remove(outgoingMessage.length() - 1);

  Serial.println(outgoingMessage);
}

void enterCommandPressed()
{
  
  Serial.print("keyboard message: \"");        
  Serial.print(outgoingMessage);  
  Serial.println("\"");    

}

void sendTextToDefaultNumber(String message) {
  Serial.print("SEND TEXT VIA KEYBOARD");
  Serial.print(message);
  Serial.println("COMPLETED");
}
`

teastainGit, 2023-05-23:
This is an I2S (ess) issue for sound, not I2C!
It is also a CLOSED issue, so few people will see it.
I think you should raise a new Issue and get others to help.
Having said that I looked at your library and header and they are declaring Wire.begin() which in 2023 is a pain in the ass. Back in the day Arduino and Adafruit tried to save you the confusion of new hobbyists trying to figure out which pins were I2C on which device, by looking at your board type. But NOW the ESP32 is all over the map.
They often get it wrong and we have to declare our I2C with our known pins, 43, 44.
I know you are doing that but the library or header may be overwriting your declaration of 43, 44 !!!
So I would try to declare Wire.begin as late as possible in the setup.
-Terry