Xinyuan-LilyGO/T-Display-S3 #65: LCD_BAT_VOLT does not work on battery.

ValeroSilva, 2022-12-12:
I'm using a 3.7 v Battery at the LCD_BAT_VOLT input but it doesn't work, only via USB. Do you have some configuration to make it work?

teastainGit, 2022-12-12:
Check out my "getting started" Repo here:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples
Bottom of the page of the setup text file, item C)

Solutions to common startup issues:
A) Unit seems "bricked" 1
   LCD does not display, unit remains dark with no white backlight:
	Add #define LCDpin 15		// just before the setup and then in setup add:
	pinMode(LCDpin, OUTPUT); 	//"triggers the LCD backlight,
	digitalWrite(LCDpin, HIGH);)	//on startup"

B) Cannot see Serial.Print on Arduino Serial Monitor
	To enable serial monitor function for debugging, you must also set 
	USB CDC On Boot "Enabled"
	in the Arduino board parameters

C) Can run on USB power but not battery connector:
	Add #define LCDpin 15		// just before the setup and then in setup add:
	pinMode(LCDpin, OUTPUT); 	//"triggers the LCD backlight, and enables battery power
	digitalWrite(LCDpin, HIGH);)	//on startup"
	
	(LilyGO variously refers to them as "LCD power on" OR "PIN_POWER_ON")
	Thanks to user RobLatour for finding this cure for the battery issue!

ValeroSilva, 2022-12-12:
"C) Can run on USB power but not battery connector:"
Problem solved! Thank you very much!

teastainGit, 2022-12-14:
> "C) Can run on USB power but not battery connector:" Problem solved! Thank you very much!
Wonderful!
When I got mine late summer, I thought I had bricked it. Brand new, broken. When I figured it out I had to share!
And it has paid me back every week from then.
Cheers, Terry

codiak6335, 2023-02-10:
So I have some more testing to do... but what I accidentally discovered:

- powered a brand new T3 from the 3.3v pin via battery and it lit up just fine... the red and green leds on the back of the board as well as the screen... the system was running the wifi scanner....    

so far so good... but oops... I need to image with micropython to run my web app so I disconnected the battery on the ping flashed it via USB... tested my app... all good

- unplugged from USB, and powered it via the 3.3v pin again... and nothing... no screen, no backside leds green or red... it just looked dead to the world....  except, the hotspot was active and functioning!!!!  so the chip is clearly being powered

I have been aware of the LCDPin for the screen background and that didn't bother me... but the backside leds leave the user with the feeling of impending doom.

closed 2023-03-13 by lewisxhe

jessedmatlock, 2023-04-24:
So, similar issue but resolutions above do not resolve it.

I can power the board with USB-C from my computer (iMac) but if I take that same cable and plug it into any of several wall power adapters (all 5v units), the screen will 'glow' softly or sometimes the loading image will show.. (from setup()) but the loop() will not run.. 

I have the LCDPin defined and set HIGH as mentioned above.. but no change..

thoughts???

lewisxhe, 2023-04-24:
Flash into the test firmware to quickly verify the problem

https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/firmware

jessedmatlock, 2023-04-24:
> Flash into the test firmware to quickly verify the problem
> 
> https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/firmware

I'm running on an iMac and no access to Windows.. any other options?

lewisxhe, 2023-04-24:
https://github.com/Xinyuan-LilyGO/LilyGo-Document/blob/master/MacFlashBin.md

jessedmatlock, 2023-04-24:
Tried but get an error, even when using SUDO 

/Library/Developer/CommandLineTools/usr/bin/python3: can't open file 'quit': [Errno 2] No such file or directory

lewisxhe, 2023-04-24:
The first command you can skip

jessedmatlock, 2023-04-24:
After consulting with a hardware engineer, he confirmed that the s3 chip implementation lacks any connections to CC1 or VConn, which is how the USB Type C often communicates current needs for upward facing ports. Causing some USB type C compliant sourcing devices to have a hard time supplying the board what it needs because of this.

So, I've opted to instead feed 5v directly to the rail and proceed that way. At least this way I can ensure there is 5v to the rail and don't have to work around an incorrect USB-C implementation

lewisxhe, 2023-04-25:
So you are using the USB TO USB interface? The USBC can’t supply power to the board, so it can’t be programmed?

jessedmatlock, 2023-04-25:
> So you are using the USB-C TO USB-C interface? The USB-C can’t supply power to the board, so it can’t be programmed?

I can power the board with the USB-C cable from my computer.. but when I use that same cable on a wall adapater the board either has a screen that barely glows, or shows my initial loading image and not the program.. sometimes the program will load but wont work... it's frustrating. 

However I think I found the issue.. and you won't believe this. ugh. I was using some of the foam from the t-display s3 case, the foam that came around it.. to hold it softly in the case (the foam was placed behind the screen and the back panel for a custom case was place behind it.. to hold it in place.) Somehow, that foam is either conductive or something.. because after removing the foam IT WORKS !!!!!!  

It was a brutal process of elimination.. I changed cables, changed wall adapters, changed sketches (which worked because I didn't close the case up with the foam in it during that test!), tried rewiring with new wires, fresh solder joints, etc.. the last piece.. the only thing different, was the foam. I thought, "No way, it can't be". But I removed it... and it works. 
🙄

lewisxhe, 2023-04-25:
Unbelievable, but now that it works it's great

lucaskatayama, 2023-04-28:
> I can power the board with the USB-C cable from my computer.. but when I use that same cable on a wall adapater the board either has a screen that barely glows, or shows my initial loading image and [...]

Ok... Changing the foam right now...