Xinyuan-LilyGO/T-Display-S3 #68: Examples Dabble Issue

buzzysale, 2022-12-17:
In the examples, particularly the NES example, the dabble app connects to ble, but does not download the pin configuration.  
"Connected". "Acquiring connected board detail" 
and then nothing happens and no detail is provided.

mmMicky, 2022-12-19:
I use Android. I can use it directly after connecting to bluetooth, there is no other prompt.

buzzysale, 2022-12-20:
How do I troubleshoot what's happening?

mmMicky, 2022-12-20:
I am also using platformIO to build. You try to upload an NES game to the data directory. Then upload to the file system to see if it is running normally.

mmMicky, 2022-12-20:
The sample doesn't have any NES games built in. Because uploading it data infringement. I can only provide methods used by the simulator.

buzzysale, 2022-12-20:
Uploading a rom is successful. That part works. I

t’s dabble that isn’t working with the esp.

buzzysale, 2022-12-20:
Okay and if that works?

mmMicky, 2022-12-20:
If it works fine, it proves that dabble software is not compatible to work on iphone.

mmMicky, 2022-12-20:
There is also a stupid but effective way. It is recommended that you find a mobile phone with Android system for testing.

closed 2023-01-06 by mmMicky