Xinyuan-LilyGO/T-Display-S3 #69: Can't get Micropython to reinstall

codiak6335, 2022-12-17:
I successfully built and installed the Micropython, boot, and partition bin files.

While playing around with the logo py example I set up a while true loop that blicked the backlight off and on with a 1-second sleep cycle.
The code worked great.... too great... as I could not break the loop and stupidly had modified boot.py

Nothing I seemed to do would break the loop.    
I did manage to use esptool to erase the flash but attempts to reinstall the 3 binaries never restored function.
Thonny can not reconnect... 

Thinking I bricked the device I shifted to Arduino IDE 2.0 and was easily able to get the C "Progress" demo running.
With a sigh of relief, I then attempted to reinstall Micropython... again, success messages... no Thonny connections
, and if I wait long enough the Progress demo restarts (like a bad penny it keeps turning up)

so the chip works, just not for MP... and I am at a loss how to progress

codiak6335, 2022-12-19:
I was able to reinstall the generic ESP32 Micropython and it seems to be working as expected.
No success in reinstalling the lilypad repo yet

teastainGit, 2022-12-19:
Good success getting Mpy stable!
If you are interested in getting your T-Display S3 running can I suggest my own repo?

https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples

I had a bad time at first and was convinced I’d bricked it, so I detailed what works for me.
My setup text is very procedural, I’m afraid, but it will get you to a running Arduino T-Display with a demo program and you can branch off and go from there.

Don’t hesitate to contact me.

Cheers !
Terry

codiak6335, 2022-12-19:
Thank Terry,  I visited your repo over the weekend, and while informative didn't resolve this issue.

I finally found a solution and have recovered with the display working.  I had upgrade esptool.py to version 4.4 something after originally getting the MCU working.   

This command line was actually failing:
esptool.py -p /dev/ttyACM0 -b 460800 --before default_reset --after no_reset --chip esp32s3  write_flash --flash_mode dio --flash_size detect --flash_freq 80m 0x0 ./build_dir/esp32s3/build-LILYGO_T-DisplayS3/bootloader/bootloader.bin 0x8000 ./build_dir/esp32s3/build-LILYGO_T-DisplayS3/partition_table/partition-table.bin 0x10000 ./build_dir/esp32s3/build-LILYGO_T-DisplayS3/micropython.bin

However, installing each bin file separately works fine:

esptool.py -p /dev/ttyACM0 -b 460800 --before default_reset --after no_reset --chip esp32s3  write_flash --flash_mode dio --flash_size detect --flash_freq 80m 0x0 ./build_dir/esp32s3/build-LILYGO_T-DisplayS3/bootloader/bootloader.bin

esptool.py -p /dev/ttyACM0 -b 460800 --before default_reset --after no_reset --chip esp32s3  write_flash --flash_mode dio --flash_size detect --flash_freq 80m 0x8000 ./build_dir/esp32s3/build-LILYGO_T-DisplayS3/partition_table/partition-table.bin

esptool.py -p /dev/ttyACM0 -b 460800 --before default_reset --after no_reset --chip esp32s3  write_flash --flash_mode dio --flash_size detect --flash_freq  0x10000 ./build_dir/esp32s3/build-LILYGO_T-DisplayS3/micropython.bin

teastainGit, 2022-12-19:
Thanks for getting back to me.
I am an Arduino C guy, and do not know python.

Cheers !
Terry Stainton

(519) 729-1047  (Call or text)
***@***.***

lewisxhe, 2023-03-13:
If you still have problems, please restart