Xinyuan-LilyGO/T-Display-S3 #70: UART1 serial

stokni, 2022-12-21:
hi, trying to get serial communication working in Arduino on 2nd UART, according to schematic GPIO17/18, but cant get any connection.
(normal serial communication via USB for debugging is working fine) Am i using the right pins? anything I am missing?

mmMicky, 2022-12-22:
Please screenshot your code.

stokni, 2022-12-22:
Very standard serial test - code confirmed working on other arduino's.
Tried Serial2 as well, but then cant upload code anymore (COM port not reconized , have to put in BOOT mode to be able to upload code) 

Tried other pins, switched TX / RX - but no luck.

```
#include "Arduino.h"

int incomingByte = 0;

void setup() {
	Serial.begin(115200);
	Serial1.begin(115200); 
	delay(1000);
	Serial.println("Serial started");

}

void loop() {
	if (Serial.available() > 0) {
		incomingByte = Serial.read();
		Serial.print("Serial received: ");
		Serial.println(incomingByte, DEC);
	}
	if (Serial1.available() > 0) {
		incomingByte = Serial1.read();
		Serial.print("Serial1 received: ");
		Serial.println(incomingByte, DEC);
	}
}

```

svdrummer, 2022-12-22:
What if you define the pins for the serial ports.

stokni, 2022-12-22:
Excellent suggestion, Serial1.begin(115200, SERIAL_8N1, 17, 18); is working.
Funny fact, seems to work on most (if not all) pins with this definition, after some quick searching read ESP32S3 can remap pins to UART/I2C etc etc? so doesnt really matter which pins I use as long as they are defined?

svdrummer, 2022-12-22:
I had the same issue. I was using the RX pin only, so....if you don't need one of the pin pairs....in my case, I didn't need the TX pin, it seems you set that pin to -1 minus one. Saves losing a pin  I am using hardware serial. Currently investigating turning if RTS and CTS. My micro resets whenever I open the serial port. Seems to be a common issue. Not much support from manufacturers.

closed 2022-12-24 by stokni