Xinyuan-LilyGO/T-Display-S3 #71: Need a T-Display-S3 board-specific definition for IDE

dapug, 2022-12-21:
T-Display-S3 is pretty much useless in PlatformIO without LilyGo contributing an official board definition for PlatformIO IDE.  Can be configured to work in Arduino IDE, ok, but a total nightmare in PlatformIO, if possible at all.

Can we run basic code, blink demos, and other more fancy demos?  Sure. Yes.  But as soon as you try and run something that needs PSRAM... nope!  You're screwed.

The "official" (current) way to use this board is by selecting  "ESP32S3 Dev Module" (ie esp32-s3-devkitc-1).  But this board is GENERIC, literally titled "Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)".  But of course, this default config is WRONG, because right off the bat, we know T-Display-S3 has 16MB Flash, and 8MB of PSRAM (granted, we have NO idea which type, Quad or Octal... lacking docs, that is an entirely other but related problem)

Configuring these things is not too hard from Arduino IDE/Tools menu, but in PlatformIO... a complete disaster, making T-Display-S3 useless for serious projects.

Add a T-Display-S3 board type for use in IDE, and don't leave us hanging with this generic thing we can't use.

mmMicky, 2022-12-22:
If you have read my example carefully you know that T-DISPLAY-S3 is perfectly suitable for platformIO. PSRAM can be used, you can directly refer to my configuration file.
https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/example/factory/platformio.ini

dapug, 2022-12-22:
When I enable PSRAM (DBOARD_HAS_PSRAM), I get **E (219) psram: PSRAM ID read error: 0x00ffffff**.  

This generally means the wrong board is selected, but considering the correct board is configured (esp32-s3-devkitc-1), it must be some other config issue.  Getting to (and understanding) all the configuration options in PlatformIO to try things is very tedious.  This is largely a fault of PlatformIO and the Espressif extensions for VSCode.  My angle here is... rather than searching the entire web for config settings, why not just bundle it all into a T-Display-S3 specific board definition?

mmMicky, 2022-12-22:
board_build.arduino.memory_type = qio_opi
In addition to enabling, OPI mode also needs to add this option. I understand your idea. If there is a T-Display-S3 model can avoid adding more options.

teastainGit, 2022-12-26:
Recommend close this Issue?

closed 2022-12-27 by mmMicky

tavdog, 2023-01-12:
I am trying to use this platformio.ini and it's not working.
The first boot does work, but then on subsequent boots I am stuck in a boot loop like the following : 
`ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420ba45e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
SHA-256 comparison failed:
Calculated: dcde8d8a4817d9bf5d5d69a7247667264e4e10ac7493514868b61f5aa6146539
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8
Tron-68B6B3231E14
mounted file system
opened config file
{"boot_num":"1","brightness":"High","config_url":"http://gt.wildc.net:8080/api/v1/gt_MAC_ADDRESS/attributes","d_opts":"p","dtype":"wind+wave","startup_message":"Maui","update_interval_minutes":"5","upsidedown":"true"}xV��?\␁
{
  "boot_num": "1",
  "brightness": "High",
  "config_url": "http://gt.wildc.net:8080/api/v1/gt_MAC_ADDRESS/attributes",
  "d_opts": "p",
  "dtype": "wind+wave",
  "startup_message": "Maui",
  "update_interval_minutes": "5",
  "upsidedown": "true"
}
parsed json_config
config_url key exists
Using config from flash
json_to_config
Default to WiFi Client
config_to_json
writing json config to flash
in Display Constructor
in tdisplay_s3 Constructor
out of Constructor
in tdisplay setup
[  1421][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[  1421][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
going for println
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021`