Xinyuan-LilyGO/T-Display-S3 #72: Schematic disagrees with pinout image

clydebarrow, 2022-12-24:
The image showing pinouts of the headers disagrees with the schematic, e.g. pin 10 and 11 of P12. The schematic shows them as connected to ESP32_S3R8 pins 49 and 50, but the diagram shows pins 44 and 43. Which is correct?
[image]
[image]

teastainGit, 2022-12-25:
From the PDF schematic:
ESP chip pin 50 is U0RXD and pin 49 is U0TXD
Header 12, pin 9 is U0RXD and pin 10 is U0TXD

From the Pinout Image:
Pinout Header Row pin 18, GPIO18 (U1 RXD)
Pinout Header Row pin 17, GPIO17 (U1 TXD)

The T-Display S3 schematic is a mess and should an embarrassment to LilyGO.
I have been asking for a better drawing for a month, or so, and no reply.

May I ask what was your concern, in terms of application?
Cheers -Terry

clydebarrow, 2022-12-25:
> May I ask what was your concern, in terms of application?

I would like to know which pins to use! Obviously I can do some testing to find the correct ones, but when bringing up code for the first time it's nice not to have to worry about whether it's my error or a documentation error.

I agree that the schematic is a mess - and Lilygo's documentation in general is woeful, which is a real barrier to using their products. Such a contrast to the Espressif documentation which is generally extremely well done.

teastainGit, 2022-12-25:
[image]

clydebarrow, 2022-12-26:
Yes, it is pins 9 and 10, not 10 and 11.  But I see where the confusion comes in. The JPG image has what look like pin numbers but are actually GPIO numbers:
[image]
So these are pins 49 and 50 on the ESP32, as per the schematic, except it labels them U0TXD and U0RXD which are the primary functions, the JPG labels them with CLK_OUT1 and CLK_OUT2 which are secondary functions.

closed 2022-12-26 by clydebarrow