Xinyuan-LilyGO/T-Display-S3 #73: USB lost identification!

RomanFoltyn, 2022-12-27:
Hello, I was looking for how to get an additional serial port and after many tests it turned out that "serial1" works on pins 16 and 17. This is not included in the Pins description in github.
However, I had another problem because the normal "Serial.print("TXT");" it did not send me a serial monitor (USB).
I thought maybe USB was connected to Serial2...
After entering the Serial2 commands into the program and sending them to Flash, the program started, but the USB lost connection to the computer. The connection is visible in the "USB-Controller" item as an unknown connection.
I don't know how to restore default now without using USB. please help because now my T-display-ESP32-S3 is useless.
Best Regards
RomanF

RomanFoltyn, 2022-12-27:
I found a solution:
You have to really press the BOOT key and then turn on the USB plug. Then the USB port is correctly recognized again.
However, this doesn't change the matter... Specifying Serial2 should not lead to a loss of connection.... if serial2 is not allowed, it should show an error when compiling!
best regards
RomanF

mmMicky, 2022-12-28:
Use the configuration in the readme, the key point is CDC ON BOOT. It will automatically simulate the USB as a serial port. The need to use other serial ports is also Serial'N'.begin(rx,tx);.

svdrummer, 2022-12-28:
I would suggest reading the "issues" section including the "closed" items. Most of the questions people ask have been covered there. Ie serial port/usb not working
Arduino settings
Defining serial ports.
etc etc

RomanFoltyn, 2022-12-29:
Hello,
of course, I checked and, for example, Serial1 sends TX data on pin 16.
My problem is that the serial should send to the USB output, which it doesn't. I did the initialization correctly.
serial monitor if I set it to 9600 it also shows ESP startup information. but the "Serial" command does not respond

teastainGit, 2022-12-29:
Have you tried my 
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples 
guide and tutorial?
If you can get this working and run the HotHead example, we can go from there to answer your 2nd serial issue
Feel free to contact me at:
 teastain@me.com !

closed 2023-03-13 by lewisxhe