Xinyuan-LilyGO/T-Display-S3 #74: Screen brightness

parahooners, 2022-12-28:
Hey - is there a simple way to adjust the screen brightness ?

teastainGit, 2022-12-28:
Well...great question...
Several hours later...
The Schematic says that GPIO38 is ESP_LCD_BLK
Which enables the clever chip "AW9364DNR"
Which dims the four LCD LED backlights here:
[image]

So, using LEDC "you can indeed" control the display brightness, proven on my T-Display S3 !!!
Add this code:

to declarations:
const int ledPin = 38;  
const int freq = 5000;
const int ledChannel = 0;
const int resolution = 8;
To setup loop:
ledcSetup(ledChannel, freq, resolution);
ledcAttachPin(ledPin, ledChannel);
To void loop:
ledcWrite(ledChannel, 25);
where 25 is percent.
Variable from 0 - 100%
Cheers, Terry

svdrummer, 2022-12-28:
I know that the information Terry wrote above works, as i have tried the same thing based on Terrys information.

parahooners, 2022-12-28:
> Well...great question... Several hours later... The Schematic says that GPIO38 is ESP_LCD_BLK Which enables the clever chip "AW9364DNR" Which dims the four LCD LED backlights here: [image]
> 
> So, using LEDC "you can indeed" control the display brightness, proven on my T-Display S3 !!! Add this code:
> 
> to declarations: const int ledPin = 38; const int freq = 5000; const int ledChannel = 0; const int resolution = 8; To void setup: ledcSetup(ledChannel, freq, resolution); ledcAttachPin(ledPin, ledChannel); To void loop: ledcWrite(ledChannel, 25); where 25 (fairly dim) is 25 of 255 max value. Variable from 0 - 255 Cheers, Terry

Thanks Terry 👍 I will give this a go

closed 2022-12-29 by parahooners