Xinyuan-LilyGO/T-Display-S3 #75: Failed to connect to ESP32-S3: No serial data received

martyee, 2022-12-30:
I got a couple new LilyGo  T-Display-S3 and can not get them to load with the Arduino IDE 2.0.3. I have updated the board manager and as much as the https://github.com/Xinyuan-LilyGO / [T-Display-S3](https://github.com/Xinyuan-LilyGO/T-Display-S3)

I have tried everything on the website and getting a fatal error occurred: Failed to connect to ESP32-S3: No serial data received 
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 2
 I'm not sure what else to try. Anyone have any ideas? Thanks in advance!!

Marty

svdrummer, 2022-12-30:
I get that sometimes. Press left hand button, while holding down, press reset button. Then release left hand button. Then send your code. Also, make sure you have a good cable.

teastainGit, 2022-12-30:
I made a start up guide for this exact board:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples
Maybe read through it and try again.
This board is VERY fussy and unforgiving of procedure
Cheers, Terry.

martyee, 2022-12-30:
Well I read through your text file. I had done a update on the standard "TFT_eSPI" Arduino library. Do I just go back in uninstall and go back to what version? Thanks!

teastainGit, 2022-12-31:
If you still have the zip file from LilyGO s GitHub just unzip it and copy the TFT_eSPI from the Lib directly and overwrite whatever file you have in your Arduino Library.
This happened to me.
I joking tell users to resist the temptation to do the Arduino Library Update!
LilyGO’s version is special because it is 8bit parallel not SPI. 

Cheers !
Terry

svdrummer, 2022-12-31:
Didn't you say it was a programming issue, which is code independent.
I also load just a "new" setup/loop only program and run that...with the reset combination above  your display won't show anything, but at least it gets your chip back. All the loading and reloading won't help. The Arduino programmer is telling you what the issue is

martyee, 2022-12-31:
I'm still having issues. I'm going to stop for the night, Thanks!

Jason2866, 2023-01-08:
> I made a start up guide for this exact board:
> https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples
> Maybe read through it and try again.
> This board is VERY fussy and unforgiving of procedure
> Cheers, Terry.

The board is not fussy, it is using a S3 which needs a 100% correct setup. Blame espressif for the unneeded trouble with OPI flash/psram.

lewisxhe, 2023-03-13:
If you still have problems, please restart

closed 2023-03-13 by lewisxhe