Xinyuan-LilyGO/T-Display-S3 #76: Suddenly unable to upload code to the device

christosfas, 2023-01-01:
I am using the esp32 s3 vesion with the touch capabilities. I have uploaded some example code successfuly the frst few times using platformio. Suddenly, whenever I try to upload an example the upload stops after uploading the first few bytes and throws an error like this:
Compressed 15008 bytes to 10328...
Writing at 0x00000000... (100 %)

A fatal error occurred: Packet content transfer stopped (received 1 bytes)
*** [upload] Error 2  

Any ideas on what I may be doing wrong and how to fix it? I have tryed multiple different cables and usb ports btw.

te-bachi, 2023-01-05:
I also have problems with flashing example projects.

I cloned [https://github.com/Xinyuan-LilyGO/T-Display-S3](https://github.com/Xinyuan-LilyGO/T-Display-S3) and uploaded example projects from the example subdirectory.

Small example projects as [example/tft](https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/example/tft) using only TFT_eSPI works as expected.
Large example projects as [example/factory](https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/example/factory) using lvgl fails during the flash process.

I use vscode and platformio and haven't changed anything in the code.

Sometimes [example/factory](https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/example/factory) fails just at the beginning
```
 Processing ESP32-S3-DevKitC-1 (platform: espressif32; board: esp32-s3-devkitc-1; framework: arduino)
[...]
PACKAGES:
 - framework-arduinoespressif32 @ 2.0.5+sha.2d6ca35
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- lvgl @ 8.3.0-dev
|-- OneButton @ 2.0.3
|-- TouchLib @ 0.0.1
|   |-- Wire @ 2.0.0
|-- WiFi @ 2.0.0
|-- Wire @ 2.0.0
Building in release mode
Retrieving maximum program size .pio\build\ESP32-S3-DevKitC-1\firmware.elf
Checking size .pio\build\ESP32-S3-DevKitC-1\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.5% (used 41092 bytes from 327680 bytes)
Flash: [====      ]  41.4% (used 1384341 bytes from 3342336 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM3
Uploading .pio\build\ESP32-S3-DevKitC-1\firmware.bin
esptool.py v4.4
Serial port COM3
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:cf:0a:e4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00162fff...
Flash params set to 0x023f
Compressed 15008 bytes to 10328...
Writing at 0x00000000... (100 %)
Wrote 15008 bytes (10328 compressed) at 0x00000000 in 0.2 seconds (effective 487.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 402.1 kbit/s)...
Hash of data verified.
[... 135 lines cut ...]
Compressed 401552 bytes to 200483...
Writing at 0x00010000... (7 %)
Writing at 0x00019ec8... (15 %)
Writing at 0x0002339a... (23 %)
Writing at 0x00031210... (30 %)
Writing at 0x0003edb4... (38 %)
Writing at 0x00044869... (46 %)
Writing at 0x0004a211... (53 %)
Writing at 0x0004f7b6... (61 %)
Writing at 0x00054cc1... (69 %)
Writing at 0x0005d050... (76 %)
Writing at 0x00065790... (84 %)
Writing at 0x0006acf1... (92 %)
Writing at 0x000707ee... (100 %)
Wrote 401552 bytes (200483 compressed) at 0x00010000 in 2.9 seconds (effective 1105.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

I use different computers (always Windows 10) and different USB cables (USB A-USB C, USB C-USB C), but it always fails.

Do you know why large example projects fail?
The flash chip is a [winbond W25Q128JV](https://www.winbond.com/hq/product/code-storage-flash-memory/serial-nor-flash/?__locale=en&partNo=W25Q128JV) with 128Mbits resp. 16MBytes... a lot of space available!

mmMicky, 2023-01-06:
I think you can go into boot mode and upload the firmware. The operation steps are to hold down the BOT button and then connect the USB or press the BOT button and click the RES button.

christosfas, 2023-01-06:
> I think you can go into boot mode and upload the firmware. The operation steps are to hold down the BOT button and then connect the USB or press the BOT button and click the RES button.

I've read this multiple times as a solution to various boot / firmware upload related issues but frankly it's on the project instructions so it's obvious and most people would have tried it already with no luck (myself included) if they knew  what they are doing at all. So I think repeating it again and again is pretty redundant.

Esptool is not anywhere near flawless especially for use with custom boards (which I have faced many similar issues with when trying to flash using esptool in other projects) and the same goes for relatively new "commercial" boards such as this lilygo product. 

Eventually, in my experience almost always the solution seems to be trying a reflash or to burn the bootloader using the Arduino IDE. This worked for me as well in this case, so I am closing the issue. Although it seems that it actually uses esptool for those tasks, and as much as I hate it for coding, it seems that the implementation / integration of some functions and programs is a lot better than platformIO.

closed 2023-01-06 by christosfas

te-bachi, 2023-01-06:
I also tried it with the Arduino IDE, and it works.
So platformio needs to fix this...