Xinyuan-LilyGO/T-Display-S3 #78: lv_demos LV_LOG_INDEV warning position y = 170 not valid (presumably 0-169)

tyeth, 2023-01-02:
Not sure if I set the resolution wrong, or if there's a little fix needed, but loading the **lv_demos** with my graph modification and then touching the bottom of screen(top?) gives a continuous warning in the serial console that the touchpoint is out of bounds.
https://github.com/tyeth/LilyGo-T-Display-S3/tree/touch_tweaks

tyeth, 2023-01-02:
```
 [Warn]     (762.506, +504117)       indev_pointer_proc: Y is 170 which is greater than ver. res    (in lv_indev.c line #368)
```

mmMicky, 2023-01-03:
Maybe the X,Y directions are swapped.

tyeth, 2023-01-03:
Wouldn't it then be the case that any value above zero would cause the error?

mmMicky, 2023-01-04:
I recommend polling to read touch sensor information. If interrupts are used, an additional layer of judgment is required. Determines the number of fingers read after an acquisition interrupt. If it is greater than 0, it is valid, otherwise it is invalid.

Jason2866, 2023-01-08:
If you want to use LVGL you may have a look at Tasmota it is fully supported.
https://github.com/Xinyuan-LilyGO/T-Display-S3/issues/18#issuecomment-1374820520

lewisxhe, 2023-03-13:
If you still have problems, please restart

closed 2023-03-13 by lewisxhe