Xinyuan-LilyGO/T-Display-S3 #79: If you are having difficulty with your new T-Display S3

teastainGit, 2023-01-03:
I made a setup guide with simple  Arduino sketch examples.
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples
No back light?
No Serial?
Backlight is ON but no image?
Won't run on battery?

Snuggert, 2023-01-24:
You helped me solve my battery problem 👍

teastainGit, 2023-01-24:
Glad to have helped!
This all happened to me!

Cheers !
Terry Stainton

***@***.***

lewisxhe, 2023-03-13:
Thank you for bringing more clear and concise examples

closed 2023-03-13 by lewisxhe

jtrumpio, 2023-10-06:
Dude thank you, it is so ridiculous there is nothing mentioned about pulling pin 15 high to get it to power on when using the little jst battery input.

teastainGit, 2023-10-07:
Thank you for your feedback!
-Terry

aw1at, 2024-04-23:
I am using MicroPython on the Bord. Setting pin 15 to high is not working. Without a new build/flash it wouldn't work right? Can I do it with a wire as well? Or would this cause a bigger issue with the backlight of the display? :(

teastainGit, 2024-04-23:
#define PIN_POWER_ON 15  // LCD and battery Power Enable
#define PIN_LCD_BL 38        // BackLight enable pin (see Dimming.txt)
You need to program these outputs "ON" in your program.
Edit your program and download it.

aw1at, 2024-04-24:
Do you have any idea how it would work with micropython? Have I set it before flashing the firmware somewhere within the make process?

teastainGit, 2024-04-24:
I do not program with Python, how would you program an LED to blink or any output to come on and go off?
These two inputs are not brought out to the external edge, but they are actual outputs just the same. They are just internal, driving the Power Enable electronic circuit on the board and the other output goes to the LCD backlight!
If you are willing to try Arduino IDE I have a complete support for this board here:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples 
-Terry

jtrumpio, 2024-04-24:
Are you also pulling pin 38 high for the backlight?

`from machine import Pin

# Define pin 15 and 38 as output pins
pin15 = Pin(15, Pin.OUT)
pin38 = Pin(38, Pin.OUT)

# Set pins 15 and 38 high
pin15.value(1)
pin38.value(1)`