Xinyuan-LilyGO/T-Display-S3 #80: time setting on Europe

Wunni73, 2023-01-05:
where is the time zone line ???

mmMicky, 2023-01-06:
Edit your time difference information here.

https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/dcc62d861dafd3a89aabd1b4dd48da5bcea1d2e6/example/factory/pin_config.h#L10

mmMicky, 2023-01-07:
What is obtained is UTC time. I'm not aware of time differences in all regions. But I know that all time differences are adjusted according to UTC time. I think users can correct the displayed time according to their own time difference information.

tyeth, 2023-01-07:
Yeah, it's just **very unclear** what to set. If you set a location then it is clear.
For example, here in the UK we use GMT, which depending on time of year has 1 hour offset or not, so the UTC offset is zero, but presumably the DST is 1hour (3600secs), but is that true at all times of the year or do you need to recompile to change the daylight savings. Also some countries switch the Daylight Saving time at different dates compared to other countries, so how the heck can that work if you don't set a location.
[image]

mmMicky, 2023-01-30:
I have an idea to modify the time difference by accessing the current IP address information after networking. In this way, the correct time can be obtained without the need for the client to modify it.

mmMicky, 2023-01-30:
I need to find a free and globally accessible API support.

tyeth, 2023-01-30:
This is a risky tactic, but probably acceptable to 95% of people. I've used a few but as you know most need api keys, this seems legit and open https://www.ipify.org/ (scroll down for usage). I'm probably glossing over my memories as I was convinced I'd set the location as a string like Europe/London, but I can't see that anywhere except some old esp issues, but it looks like the https://github.com/nayarsystems/posix_tz_db/blob/master/zones.csv zones file was what I used for lookup, combined with setenv and tzset like https://randomnerdtutorials.com/esp32-ntp-timezones-daylight-saving/
One user even keeps the whole zones file in spiffs and parses if required (user change)...madness but works.

mmMicky, 2023-01-31:
I updated the method to get the timezone automatically. In theory, users anywhere can access local time. But batch verification is missing. I tested it locally and it works fine.

tyeth, 2023-01-31:
I like it, works well. I've added a custom override with simple #ifdef check, not sure you want it, but stuck in a pull request in case you do.

lewisxhe, 2023-03-13:
If you have any questions, please open it again

closed 2023-03-13 by lewisxhe