Xinyuan-LilyGO/T-Display-S3 #81: problems using SD card reader (SPI) with the touch version

Hans0lo, 2023-01-05:
Hi,
I try to connect a SD reader via SPI to the T-Display-S3 touch version - sadly without success.
From reading other comments/websites it seems I can freely define the pins for a secondary (HSPI) SPI bus.
This is what I tried:
-----------------------------------------------
`#include "FS.h"
#include "SD.h"
#include "SPI.h"

SPIClass spiSD(HSPI);
#define SDSPEED 27000000

//  SPI port #2:  SD Card Adapter
#define SD_CLK 21 //yellow
#define SD_MISO 18 //brown
#define SD_MOSI 17 //orange
#define SD_CS 16 //green`

in setup I have:
---------------------------------
`  Serial.begin(9600);

  spiSD.begin(SD_CLK, SD_MISO, SD_MOSI, SD_CS);  //SCK,MISO,MOSI,cs
  
  if (!SD.begin(SD_CS, spiSD, SDSPEED)) {
    Serial.println("Card Mount Failed!");
    return;
  } else {
    Serial.println("Card Mount Successful!");`

but all I get is "Card Mount failed" - I have multiple reader modules, but I get the same result with all of them, so i assume the issue is with my setup/pins/code...

any help would be highly appreciated - thanks in advance  :)

mmMicky, 2023-01-06:
First, you need to ensure that the file system format of your SD card is FAT32. Secondly, in addition to the SPI communication method. There is also a SD_MMC 1bit method. This method only needs to use 3 data lines and has a huge improvement in timing.
The method of use is as follows
SD_MMC.setPin(x,x,x);
SD_MMC.begin("/sdcard",true,true);

Hans0lo, 2023-01-06:
Hi @mmMicky & thanks for your help.

my file system already was FAT32, so this was not the issue.

And while I may look into the MMC 1-bit method, I still would like to get the SPI mode working as (for other projects) I also want to use other SPI peripherals with this T-Display-S3 board, not only SD card...

I therefore rephrase my questions:

1. can I use a second (HSPI) bus with this display?

2. If yes, can I use this port setup?  

#define SD_CLK 21 //yellow
#define SD_MISO 18 //brown
#define SD_MOSI 17 //orange
#define SD_CS 16 //green

3. is there any code example for using a second SPI bus with this board?

thnx a lot for your help!

Hans0lo, 2023-01-06:
I found the solution on my own with some help of Google.
In the end I only had to add one more line of code to make it work (switching the CS Pin to output...):
```
pinMode(SD_CS,OUTPUT);
```

so the complete code-part looks like this:
```
void setup() {

  Serial.begin(9600);

  spiSD.begin(SD_CLK, SD_MISO, SD_MOSI, SD_CS);  //SCK,MISO,MOSI,cs
  pinMode(SD_CS,OUTPUT);

  if (!SD.begin(SD_CS, spiSD, SDSPEED)) {
    Serial.println("Card Mount Failed!");
    return;
  } else {
    Serial.println("Card Mount Successful!");
  }
}
```

mmMicky, 2023-01-07:
Congrats on finding the right solution and sharing it. This is also required when using SD_MMC.
pinMode(cs, output);
digitalWrite(cs, 1);

closed 2023-01-07 by Hans0lo